rogerclarkmelbourne/Arduino_STM32 #20: EEPROOM can not compile

taoyuan, 2015-01-05:
Include "EEPROM.h" into ino project, throw the error when compile:
```
.../flash_stm32.c:3:19: fatal error: flash.h: No such file or directory
 #include "flash.h"
```

rogerclarkmelbourne, 2015-01-05:
Thanks for letting me know.

The library is was written for LibMaple, but it looks like things have moved on since it was written.

I will try to track down the original author of the file.

I took a quick look to see if I could find a more up to date version, and AreoQuad's version of libmaple has a version of these files

See. https://github.com/AeroQuad/AeroQuad/tree/master/AeroQuad32/MapleCompatibility

Which may work better than the version I found and included in the repo.

taoyuan, 2015-01-05:
It's helpful, thanks.

avikde, 2015-01-05:
Hi, this isn't strictly related to the compilation issue but I've been wanting for a while to see if it's possible to have EEPROM data persistent across reprogramming (usually programming does an extended erase, which erases all of the flash memory), and I think I've finally figured it out. I [wrote about it here](http://avikde.me/saving-flash-eeprom-data-when-bootloading/), feel free to take a look and see if it's helpful!

BTW, it's still only tested on F373, but I hope to be able to test [koduino](https://github.com/avikde/koduino) on F4, and then F1 targets soon-ish (in the next month or so, once I have some hardware and some time) and hopefully we can combine efforts somehow, or at the very least, share some libraries.

rogerclarkmelbourne, 2015-01-05:
Thanks for the link to your post.

I have not investigated it, but the STM uploader, windows exe, has a feature to only erase the banks of memory which need to be reprogrammed.

This may also be what the Maple bootloader does when you upload via DFU.

When I briefly looked th the EEPROM code, it appeared to use the very end of the flash memory, e.g the last bank, or possibly the last 2 banks etc.

Perhaps taoyuan will have some success with the derivative code from the AeroQuad repo

rogerclarkmelbourne, 2015-07-04:
AFIK EEPROM lib is now working (and has been for some time)

closed 2015-07-04 by rogerclarkmelbourne