rogerclarkmelbourne/Arduino_STM32 #22: Maple mini generic - OSX

madias123, 2015-01-17:
Hello,
is there a way to implement the "upload router" for OSX, so mac user can use the generic board?

Thank you for that great project!
regards
Matthias

rogerclarkmelbourne, 2015-01-17:
Matthias

I'll investigate what need to be added.

madias123, 2015-01-17:
Dear Roger,
thank you for that!
I could perform  tests on the following Boards I own under OSX: Maple mini (remake) for serial connection (usb works perfect) and  a Nucleo STM32F103 for testing the ST-Link upload (before I've to write board definitions). By the way, it would be a big think supporting the Nucleo devices, the STM32F103 should be a easy thing :)

rogerclarkmelbourne, 2015-01-18:
Matthias

STLink upload to Nucleo should be possible, I have tested STLink to F103 on the PC.

However the problem with the STLink is that a different method to send serial debug messages would be needed, and I think its only one direction  i.e debug messages from STLink to the PC / OSX

madias123, 2015-02-06:
this issue can be closed, everything is fixed! :)

closed 2015-02-06 by madias123