rogerclarkmelbourne/Arduino_STM32 #23: Is there a plan to implement pins compatibility macros ?

taoyuan, 2015-02-06:
Many libs using pins macros such as [Adafruit_ILI9341](https://github.com/adafruit/adafruit_ILI9341). 

Here is the pins compatibility macros needed.

```c
#define digitalPinToPort(pin)
#define digitalPinToBitMask(pin)
#define portOutputRegister(pin)
#define portSetRegister(pin) 
#define portClearRegister(pin) 
#define portToggleRegister(pin) 
#define portInputRegister(pin)
#define portModeRegister(pin)  
#define portConfigRegister(pin) 
```

madias123, 2015-02-06:
there is a (problematic!!!!) dummy in every variant/maplexyz/variant.h 
timschuerewegen released a VERY RECOMMENDED patch (not only for SPI but  far more compatibility and bugfixes) in the arduino forum thread #1361.
With this patch, the following lines are added to (some!!!) variant.h:

#define digitalPinToPort(P)        ( PIN_MAP[P].gpio_device )
#define digitalPinToBitMask(P)     ( BIT(PIN_MAP[P].gpio_bit) )
#define portOutputRegister(port)   ( &(port->regs->ODR) )
#define portInputRegister(port)    ( &(port->regs->IDR) )

In my opinion, I would enter these changes globally, because they are not device dependent.

rogerclarkmelbourne, 2015-02-06:
I'm not sure that simply implementing these direct hardware access macros is going to be a one size fits all solution 

Alexey (@hiddenpilot) did some to speed up access to hardware with new API commands for digitalWriteFast etc, which I put into a branch of the repo, but this isn't entirely the same as those macros

The problem is that the way the STM32 port registers work isn't consistent with the way the AVR port registers work, so as well as defining these macros, I think each library needs a custom implementation of how to use the macros.

Rather than writing these macros to just work with one library, we need to consider all the libraries that use direct access to the hardware e.g. OneWire 

Otherwise we will create a solution which only works for one library

I agree however that the macros need to be in the core. But having broken the core several times, by applying fixes that people have sent me, I have become cautious about changing things that I can't easily test, or changes without details

The best way to manage change is to use git and for anyone to clone the repo, make changes, get a load of people to test them, and then do a pull request back into this repo

rogerclarkmelbourne, 2015-02-09:
The maple mini variant now has definitions for ..

#define digitalPinToPort(P) ( PIN_MAP[P].gpio_device )
#define digitalPinToBitMask(P) ( BIT(PIN_MAP[P].gpio_bit) )
#define portOutputRegister(port) ( &(port->regs->ODR) )
#define portInputRegister(port) ( &(port->regs->IDR) )

Which were added by Tim as part of his SPI patch, but I think are actually used by the ILI9341  library (LCD display)

Ideally Tim should have added this as part of a different patch, but its done now, and people are using it, so I'll leave it.

I'm not sure why Tim didn't stick them in the core somewhere. 

I'll probably move them to somewhere like wirish.h or some other file that is included all the time.

At the moment I don't think I have any way to test these macros, as I don't have that display, so can't use that Library (though I have now ordered a LCD display from eBay).

I think that the OneWire library may use some or all of those macros, so if I get time I'll see if I can test it, as I have some OneWire thermometer DS18B20's 

Re: the Other macros not implemented by Tim.

#define portSetRegister(pin) 
#define portClearRegister(pin) 
#define portToggleRegister(pin) 
#define portModeRegister(pin)  
#define portConfigRegister(pin) 

It shouldn't be too hard to write macros for these, as the SetRegister is a fixed offset from the other register addresses (same for the Clear and Toggle), But I'll need to check for Mode and Config, and with Mode and Config I can't see how any AVR code would work without modification, as its unlikely the the STM mode register operates in the same way as the STM32 mode register etc

rogerclarkmelbourne, 2015-02-15:
@taoyuan 

Looking at the list of pins compatibility macros you initially posted, I've just realized that some of these are not available on either the AVR or SAM variants of the Arduino.

e.g. Set and Reset registers are not in either as far as I can see.

This appears to be the list defined for AVR

#define digitalPinToPort(P) ( pgm_read_byte( digital_pin_to_port_PGM + (P) ) )
#define digitalPinToBitMask(P) ( pgm_read_byte( digital_pin_to_bit_mask_PGM + (P) ) )
#define digitalPinToTimer(P) ( pgm_read_byte( digital_pin_to_timer_PGM + (P) ) )
#define analogInPinToBit(P) (P)
#define portOutputRegister(P) ( (volatile uint8_t *)( pgm_read_word( port_to_output_PGM + (P))) )
#define portInputRegister(P) ( (volatile uint8_t *)( pgm_read_word( port_to_input_PGM + (P))) )
#define portModeRegister(P) ( (volatile uint8_t *)( pgm_read_word( port_to_mode_PGM + (P))) )

and SAM its
#define digitalPinToPort(P)        ( g_APinDescription[P].pPort )
#define digitalPinToBitMask(P)     ( g_APinDescription[P].ulPin )
//#define analogInPinToBit(P)        ( )
#define portOutputRegister(port)   ( &(port->PIO_ODSR) )
#define portInputRegister(port)    ( &(port->PIO_PDSR) )
#define digitalPinHasPWM(P)        ( g_APinDescription[P].ulPWMChannel != NOT_ON_PWM || g_APinDescription[P].ulTCChannel != NOT_ON_TIMER )

Without knowing why you want these non-arduino macros for, its not even practical to even test them.

I want to get OneWire working, but I think I'll do it within the confines of just implementing the core Arduino macros, i.e those available on AVR.

taoyuan, 2015-02-28:
@rogerclarkmelbourne 

Sorry for such a long time there were no replies. I just got back from the Chinese new year holiday.

It's my mistake about the list of pins compatibility macros I initially posted. I rechecked the official list, it is exactly as same as you posted.

rogerclarkmelbourne, 2015-02-28:
Hi taoyuan 

Belayed Happy New Year ;--)

No worries.

I have put all the normal Pin Compatibility macros into the variant.h for all supported variants. I may be be able to move them into the core some time, but I'm not sure yet.

I have ported the OneWire library, but the way that OneWire used the pin macros could not be completely replicated for STM32 for various reasons 

But its OK for some libraries.

Anyway, the project is still moving forwards and I have additions and bug fixes quite often from various contributors, which is good for everyone

Cheers

Roger

closed 2015-02-28 by rogerclarkmelbourne

pinchies, 2019-03-27:
We're also missing the digitalPinHasPWM macro for Marlin 3D Printer firmware on STM32F1.

stevstrong, 2019-03-27:
That example is based on HAL, this is libmaple core.

pinchies, 2019-03-27:
@stevstrong ... but I didn’t know there was a difference. 😂