rogerclarkmelbourne/Arduino_STM32 #24: implement timschuerewegen patch on thread #1361 (arduino forum)

madias123, 2015-02-06:
I have examined the patch (line for line) and have come to the conclusion that it is a great enrichment for compatibility reasons and improvement
some examples:
include the "variants.h" into  Arduino.h
include <wirish_constants.h> into wirish.h
adds some constants and  typedefs (typedef uint32_t uint32;)
a complete overhaul for the SPI class, SPI now works as it should.

rogerclarkmelbourne, 2015-02-06:
madias

Can you send me a link to the patch / diff file you applied,

I'll try re-applying the patch and see if I get the same errors.

I'm not sure what happened, as I applied the patch on my mac, and it seemed OK, but the next day when I pulled the latest version of the repo onto my PC, it would not compile the "bare minimum"

Can you confirm that you don't get errors while compiling sketches that don't include SPI ?

Thanks.

madias123, 2015-02-07:
Dear Roger,
I took the same link as you: http://forum.arduino.cc/index.php?action=dlattach;topic=265904.0;attach=112246
But I only applied  the patch on OSX and copied the whole STM32-Arduino folder to windows, because I've no working patch/diff.exe on my windooze system, had permanent troubles with them. 
On both machines I didn't got any error, even with or without including SPI.

rogerclarkmelbourne, 2015-02-08:
Matthias

I figured out what went wrong last time I applied the patch.

Tim didnt just patch some existing files, he created a new file called wirish_constants.h, but I didnt realise that he'd done this.

Consequentially, I didnt use git add to add the new file, I just committed and pushed.

On OSX its not as obvious as on the PC, that there was a new file, because on the PC the Git Shell has a colour coded status.

Anyway. It should now be OK

But...

I've also noticed that Tim only changed one variant, the maple mini ...

So it looks like I'll need to change all the other variants to make them compatible ;-(

But I'll look at that tomorrow.

rogerclarkmelbourne, 2015-02-13:
Patch has now been applied

closed 2015-02-13 by rogerclarkmelbourne