rogerclarkmelbourne/Arduino_STM32 #25: Files missing for Maple

erniberni, 2015-02-15:
Hi, 
I installed Arduino 1.6.0 and the Arduino_STM32-master files like explained. When compiling programms for Maple Rev 3 to Flash I got the error 
arduino-1.6.0/hardware/Arduino_STM32/STM32F1XX/cores/maple/Arduino.h:42:21: fatal error: variant.h: No such file or directory
 #include "variant.h"
After copiing the files variant.h and pins_arduino.h from directory 
~/arduino-1.6.0/hardware/Arduino_STM32/STM32F1XX/variants/maple_mini   to
~/arduino-1.6.0/hardware/Arduino_STM32/STM32F1XX/variants/maple
it compiled without errors.

Regards
Reinhard

rogerclarkmelbourne, 2015-02-15:
Thanks for letting me know.

I think I must have accidentally moved it, when making a new board type, and then deleted the new board , hence I ended up deleting variant.h

I will re instate it from the previous commit ASAP

rogerclarkmelbourne, 2015-02-15:
Reinhard

OK.
Fixed.

Long story.
It looks like a patch for SPI from timschuerewegen, added and include of variant.h in Arduino.h, but there has never been a variant.h in the maple variant folder. Tim should have added this for all variants that didn't already have this file.

Unfortunately, Tim did not supply any documentation about what he'd changed or why, and I obviously didn't do enough testing after applying his patch / diff file

Anyway, your  suggestion was correct. Using the Maple mini file is fine.

Actually, I'm to probably move the pin macro's to Arduino.h so the variant.h is probably not needed, however, I'm still working on some pin macro's that are needed for some libraries, which Tim didn't create. So I want to get it all working before doing a big update.

Thanks again for letting me know the issue.

closed 2015-02-15 by rogerclarkmelbourne