rogerclarkmelbourne/Arduino_STM32 #27: Pin compatibility macro's won't work in global constructors

rogerclarkmelbourne, 2015-02-17:
NOTE.

Any code which uses the Pin compatibility macro's in its constructor - when called globally, e.g. like the OneWire example does.

e.g.

 OneWire  ds(PA4);  // on pin 10 (a 4.7K resistor is necessary)

 void setup(void) {
   Serial.begin(115200);
 }

Will not work.

This is because the PIN_MAP array that contains the data used by the pin compatibility macros. for example

#define digitalPinToPort(P)        ( PIN_MAP[P].gpio_device )

is also effectively constructed at the same time as any global constructors e..g

extern const stm32_pin_info PIN_MAP[BOARD_NR_GPIO_PINS] = {

    /* Arduino-like header, right connectors */

    PMAP_ROW(GPIOA,   3,   NULL,  0, ADC1,    3), /* D0/PA3 */

Hence its impossible to guarantee that PIN_MAP is initialized before other globally initialized items.

rogerclarkmelbourne, 2015-05-03:
Note. 

@victorpv  has looked at putting the PIN_MAP into the flash memory segment, if we can get the compiler to do this, then it should be available without initialisation, and therefore we should no longer have this issue.

However I await victors work with this.

madias123, 2015-05-29:
Is this issue fixed meanwhile?

rogerclarkmelbourne, 2015-05-29:
Long story ;-)

Victor managed to move the PIN MAP into flash and also modified the initialization code so that the core variables were initialized first.
I put all the code into a branch

https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/PIN_MAP-in-FLASH

And it worked fine for me

However there was resistance from Ray as he though there was a speed issue as Flash is slower than RAM, and also RIck didn't like it either (I can't remember his reasons)

Personally, I don't see a problem with it, and I think when the dust settles I would like to merge the branch back into the master version of the repo.

I'm not sure if anyone did any real timing tests to confirm that it really would be slower for intensive GPIO code if the PIN MAP was in flash.
Even if it is a bit slower, I can easy speed digitalWrite etc up a tiny bit because it has a test for the pin number being in range, in the top of most GPIO functions, - which IMHO is pointless, because the code may as well crash as do nothing, if you have tried to use a pin number that doesnt exist.

And as I only use the real pin numbers e.g. PC13 its impossible to use a pin that is not available (as there would not be an enum for it ) - if you see what I mean.

victorpv, 2015-05-29:
We should be able to apply the fix for the constructors even if we keep PIN_MAP in RAM.
The solution for the constructors issue didn't come from moving PIN_MAP to flash, but rather from setting the highest priority within the constructor's group to the premain() routine that sets the ports.
The details in this post:
http://stm32duino.com/viewtopic.php?f=16&t=91&p=682&hilit=constructor#p682

I think we should apply that one change to the global branch.
Regarding the pin access speed when in flash, I did some tests measuring how fast I could toggle a pin in one case and the other and there was something like 10% difference in speed or something like that. It's in the same thread.

The problem is that the thread got messed with different issues and I believe most people lost track that there were different unrelated issues being discussed.

rogerclarkmelbourne, 2015-05-29:
Victor

Ok. I need to do a staged release.

I have pending releases for the whole archive thing, and Matthias's remapping of Serial

But I will try to push them all through at the weekend and wait for people to scream at me for breaking the repo ;-)