rogerclarkmelbourne/Arduino_STM32 #29: upload to ram?

snelweg, 2015-02-20:
Hi,

Would be nice to be able to upload code to RAM. Any plans to put that in too?
(and thanks! works great)

rogerclarkmelbourne, 2015-02-20:
The option to upload to RAM was commented out from the boards.txt file a while ago, because even the bare minimum sketch would no longer fit.

We made a recent change to save 5k, so it may fit in again now, but the space for a sketch would be minimal.

Also if the link options are changed to use newlibnano instead of newlib would make the binary smaller, but with possible incompatibility issues.

I'm not sure why you would want to upload to RAM, can you elaborate ?

rogerclarkmelbourne, 2015-02-23:
@snelweg

Note. I just had to revert a change suggested by @evq, which reduced the binary size, because without the whole-archive flag in platforms.txt, there were issues with the Hardware timers 

The binary size is now too large to practically fit in RAM on most devices.

If you really want to try uploading to RAM on your board, just uncomment this section in boards.txt

```
## F103CB USB to RAM upload
## -------------------------
#nano.menu.cpu_upload_menu.cbUsbRAM=F103CB(128k) USB->RAM
#nano.menu.cpu_upload_menu.cbUsbRAM.build.ldscript=ld/ram.ld
#nano.menu.cpu_upload_menu.cbUsbRAM.build.cpu_flags=-DMCU_STM32F103CB -DBOOTLOADER_maple 
#nano.menu.cpu_upload_menu.cbUsbRAM.build.vect=VECT_TAB_RAM
#
#nano.menu.cpu_upload_menu.cbUsbRAM.upload.maximum_size=17000
#nano.menu.cpu_upload_menu.cbUsbRAM.upload.ram.maximum_size=17000
#nano.menu.cpu_upload_menu.cbUsbRAM.upload.flash.maximum_size=108000
#nano.menu.cpu_upload_menu.cbUsbRAM.upload.altID=0
#nano.menu.cpu_upload_menu.cbUsbRAM.upload.protocol=maple_dfu
```

closed 2015-02-23 by rogerclarkmelbourne