rogerclarkmelbourne/Arduino_STM32 #30: No serial2, serial3 when chossing "STM32 to Flash - No bootloader"

thomasfla, 2015-02-23:
It seems there is a problem with the serial2, serial3 definitions when chossing "STM32 to Flash - No bootloader"

error: 'Serial3' was not declared in this scope

rogerclarkmelbourne, 2015-02-23:
There isn't a Serial 3 for that board type.

Mapping for the non Maple device is for HardwareSerial 1 to be mapped to "Serial", 
HW Serial 2 -> Serial 1, HW Serial 3 -> Serial 2 

Spec for even the STM32F104R series is

"Up to 3 USARTs (ISO 7816 interface, LIN, IrDA capability, modem control)"
 
What processor do you have on your board i.e do you have STM32F103Z series which has more usarts 
"Up to 5 USARTs (ISO 7816 interface, LIN, IrDA capability, modem control)"

Currently support is really only for STM32F103C and STM32F103R series, but it don't think it would be hard to add support for Z series etc

thomasfla, 2015-02-23:
my mistake! I did not notice what the Serial indexing was deferent from the HW. I use STM32F103C8T6 on this board:
http://www.buyincoins.com/item/45180.html#.VOsFrXWG-kA

It works well using the Serial2.

Nice day.

rogerclarkmelbourne, 2015-02-23:
No worries

The mapping was changed for compatibility with all existing Arduino code, i.e which uses Serial as the first serial port (even on Mega2560 etc)

I have a very similar board, and it works fine.

closed 2015-02-23 by rogerclarkmelbourne

Maceraperest, 2016-07-24:
"STM32 to Flash - No bootloader" option does not appear

rogerclarkmelbourne, 2016-07-25:
@Maceraperest 

I can't see how your problem relates to this issue.

Either create a new issue with full details of precisely what you did to replicate 
or if you need support register with www.stm32duino.com

PS. I suspect your issue is that you choose a board option like "Maple mini" which does not have No Bootloader as an option.

Not all boards support all upload options

Maceraperest, 2016-07-25:
Problem;
1. Arduino Version 1.6.9 lower version 1.6.4  I installed
2. Boards Manager Arduino Sam board 32 Bit  I installed

Fix the problem. Thank you for your interest