rogerclarkmelbourne/Arduino_STM32 #32: STM32F1XX/boards.txt microduino identifier

apla, 2015-02-24:
microduino board identifier is `nano`. There is no issue with Arduino IDE, but `nano` name is shared with Arduino Nano board.

Month ago I released [cuwire](http://apla.github.io/cuwire/) — IDE for microcontrollers, based on Arduino 1.5 hardware specification. And command line tool can assume board vendor (like `arduino` or `Arduino_STM32`) by board name. After installation of STM32 files `nano` shortcut of board name is not working anymore.

I can make pull request for that change, but want to discuss before. Probably you have a good reason to name `microduino` as `nano`

rogerclarkmelbourne, 2015-02-24:
@apla 

I didnt personally write the mcrodino stuff, it was a submission to me (I don't have any microdino hardware either)

But I know what you mean.

the 'nano' name use started when another contributor Alexey (@HiddenPilot), added the "Generic mini maple' boards last year.

He had issues with the internal names in the boards.txt and started calling his stuff 'nano' to get it to work, and the contributor that did the microdino stuff seems to have copied Alexy's mistake.

I'd not looked in detail at how the microdino section was identified and I agree that this is bad practice and should be changed (as should the  Generic maple mini  stuff)

PS. It was later determined that Alexey's issues with identifier name were because of the use of underscores or dashes, I can't remember which, but one or both of these doesn't work correctly if used as a board identifier

Anyway, if you want to clone and do a pull request, that would make it easier for me to update.

Re: http://apla.github.io/cuwire/

OK. Sounds interesting. I know there are several other competing IDE's and some users of my repo are trying to get it to work with Eclipse
 
If I get time I will download cuwire

rogerclarkmelbourne, 2015-02-25:
I just took a look at boards.txt and the microduino identifier in boards.txt is not nano is microduino32 not nano

```
##############################################################

microduino32_flash.name=Microduino Core STM32 to Flash

microduino32_flash.upload.tool=maple_upload
microduino32_flash.upload.protocol=maple_dfu
microduino32_flash.upload.maximum_size=108000
microduino32_flash.upload.use_1200bps_touch=false
microduino32_flash.upload.file_type=bin
microduino32_flash.upload.ram.maximum_size=17000
microduino32_flash.upload.flash.maximum_size=108000
microduino32_flash.upload.usbID=1EAF:0003
microduino32_flash.upload.altID=1
microduino32_flash.upload.auto_reset=true

microduino32_flash.build.mcu=cortex-m3
microduino32_flash.build.f_cpu=72000000L
microduino32_flash.build.core=maple
microduino32_flash.build.extra_flags=-DMCU_STM32F103CB -mthumb -DSTM32_MEDIUM_DENSITY -DBOOTLOADER_maple -march=armv7-m -D__STM32F1XX__ 
microduino32_flash.build.ldscript=ld/flash.ld
microduino32_flash.build.variant=microduino
microduino32_flash.build.variant_system_lib=libmaple.a
microduino32_flash.build.vect=VECT_TAB_FLASH
microduino32_flash.build.density=STM32_MEDIUM_DENSITY
# FIXME!
microduino32_flash.build.error_led_port=GPIOB
microduino32_flash.build.error_led_pin=1
microduino32_flash.build.gcc_ver=gcc-arm-none-eabi-4.8.3-2014q1```

Do you mean the generic board is set to nano which conflicts with nano 

```nano.name= Maple mini generic```

apla, 2015-02-25:
my mistake, sorry. yes, generic maple mini board have nano identifier

rogerclarkmelbourne, 2015-02-25:
Fixed !

rogerclarkmelbourne, 2015-02-25:
PS. Made a load of other changes today

remove loads of redundant / duplicate  .c and .s files from the system folder 

I also need to rename the STM32F1XX folder as we now have STM32F3 and STM32F4, so the old STM32F1XX folder name is not compliant with the naming convention on the new folders

However this should not cause any changes to the build process. Its purely cosmetic in the file system

rogerclarkmelbourne, 2015-02-26:
I presume this is OK now.  

Let me know if there are any other problems

closed 2015-02-26 by rogerclarkmelbourne

apla, 2015-02-26:
Works great! Thank you!

rogerclarkmelbourne, 2015-02-26:
No worries.