rogerclarkmelbourne/Arduino_STM32 #39: Arduino IDE 1.6.2 is incompatible

rogerclarkmelbourne, 2015-03-31:
The Arduino IDE made some major changes to the IDE file structure in version 1.6.2 which cause it to not work with this repo

At the moment I suggest that everyone stay with 1.6.1 as this is known to work.

The reason is that the IDE team have completely restructured the files that are installed, to make it modular, and 1.6.2 no longer installs the Due board and hence no longer installs the ARM compiler by default

It is possible to "Install" the Due and to get the ARM compiler, but the ARM compiler is installed in a completely different directory, hence the path specified for the compiler in platforms.txt is no longer correct.

There is a work around for this. However if I change platform.txt it breaks the repo for all previous versions :-(

I'm trying get the Arduino IDE team to add a feature so that third party boards can be added by the Boards Manager feature.
So watch this space...

apla, 2015-04-01:
Some links:

 * [Discussion on Google Groups](https://groups.google.com/a/arduino.cc/forum/#!topic/developers/E9ifjXwV2w0)
 * [Topic on Arduino Forum](http://forum.arduino.cc/index.php?topic=311851.0)
 * [Java code for library index url](https://github.com/arduino/Arduino/blob/3788128385d0ca21f4652246f205f838d7c4cf54/app/src/cc/arduino/contributions/libraries/ui/LibraryInstaller.java#L50) — you can override it with java -DLIBRARY_INDEX_URL=<new url>
 * [Java code for packages url](https://github.com/arduino/Arduino/blob/bcc88fdd9ef01b1b9a63f3051e034b6a5b26c1f6/arduino-core/src/cc/arduino/contributions/packages/ContributionInstaller.java#L60) — you can override it with java -DPACKAGE_INDEX_URL=<new url>

rogerclarkmelbourne, 2015-04-01:
@apla 
Thanks

I think I figured out how to do this.

Open  arduino.l4j.ini   in the IDE folder and add that define to the end e,g,

-Xms128M
-Xmx512M
-DPACKAGE_INDEX_URL=http://www.rogerclark.net/downloads/package_index.json

I need to reboot to double check, but I think it works !

rogerclarkmelbourne, 2015-04-01:
OK.

Yes. It works.

I just replaced the Arduino name with my name RogerClark, to test

I guess I can now work on how to dynamically add more packages.

I could write a PHP file that reads in the file from arduino.cc decodes the json, then adds more items to the packages array (top level)

However changing  arduino.l4j. is not that easy for anyone, as it required admin privileges on windows

apla, 2015-04-01:
On Mac this file is `Arduino.app/Contents/Info.plist`

Isaac96, 2015-04-07:
Hi guys I found a way to get the compiler path to work in 1.6.3 and 1.6.2
You go to your hardware folder, open STM32, then open STM32F1.
Edit your "compiler.path" variable to {runtime.tools.arm-none-eabi-gcc.path}/bin/.
Do the same for STM32F3 and F4.
This doesn't allow gcc 4.9, which I couldn't  find in the system(I deleted it from boards.txt)
I don't have a Maple yet  =(  but that's my next plan.
This edit makes compile simple  =)

rogerclarkmelbourne, 2015-04-08:
@isaac96

Does this work with 1.6.1 and 1.6.0?

apla, 2015-04-08:
Arduino IDE development decisions is complete chaos. In 1.5.x-1.6.1 they slowly moving towards 3rd party hardware. In 1.6.2 new hardware folder is added and built-in hardware is divided into hardware and tools (right move, but without discussion and no support for 3rd party). In 1.6.3 avr core and tools is returned into app contents (WHAT?), and configuration replacements created for _legacy_ cores.

If your board config contains 
```
compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/
```
after load, it will be interpreted as:
```
compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/
```

without any platform.txt modifications

replacements: https://github.com/arduino/Arduino/blob/master/hardware/platform.keys.rewrite.txt

Isaac96, 2015-04-08:
@rogerclarkmelbourne  No it doesn't work with 1.6.1 (just tried it 30 seconds ago).
About 1.6.0, I am downloading it right now to try.
I'll update in a few minutes

rogerclarkmelbourne, 2015-04-08:
@Isaac96 

Dont bother to download 1.6.0, if your change doesnt work in 1.6.1 it wont work on earlier versions

The issue is the IDE team fundamentally changed the storage location in 1.6.2 and they didnt provide any compatibility for existing users who use the ARM compiler - which was part of the IDE from 1.5.0 to 1.6.1

The ARM compiler is now only installed if you install the Due and also, the compiler is in a totally different location

There are also issues about where the compiler has been stored, as its in breach of Microsoft guidelines on windows and causes Linux users issues as well

However the IDE team seem hell bent on carrying on with whatever they want to do regardless of backwards compatibiliy to third party hardware

The only long term solution may be for me to add the whole of the ARM compiler to the repo

Isaac96, 2015-04-08:
@rogerclarkmelbourne 
Dang!
I just downloaded 1.6.0 and it didn't work(of course).

Isaac96, 2015-04-08:
Also ,the ARM compiler is 250 megs (on Windows 7), and that's going to bloat the repo.
Currently, the entire STM32 folder is 33 megs. You are going to increase the size by 750% (I used a calculator for that :+1:

rogerclarkmelbourne, 2015-04-08:
Yes. I know. And I'd need to have the OSX and Linux and Linux64 binaries as well

At the moment I'm just telling everyone to use 1.6.1

Or use Arduino.org's version 1.7 which still includes the ARM compilor by default

madias123, 2015-04-09:
I do not recommend the Arduino.org version 1.7, because it seemed to be an old fork (1.5.x or 1.6.0, maybe 1.6.1?)  with an irritating version number. I'm not sure, but my feeling says to me, to avoid arduino.org
https://github.com/arduino-org/Arduino/issues/2

rogerclarkmelbourne, 2015-04-09:
Matthias

I know several people on the forum including Ray, have tested 1.7 from Arduino.org and are using it now in preference

I'm using 1.6.1 still from arduino.cc

madias123, 2015-04-09:
I know, but arduino.org had started to bring their internal "war" to the users and these idiots are intentionally trying to confuse the user with such things like "we are the real arduino" and this misleading version number. So it should be a statement not to recommend any software from them.

MauroMombelli, 2015-04-09:
@rogerclarkmelbourne please can you join the discussion on arduino.cc mailing list? the new system will certainly cause issue at first, but basically the idea is to dynamically download in install support for board different from "classic" AVR, this should help a lot adding new hardware AFAIK as it will be like adding a repo and installing a plugin.
There is still some issue and things will still change a bit, so obliviously the opinion of a 3° part core developer as you is invaluable.

ffissore, 2015-04-09:
hi @rogerclarkmelbourne I'm sorry to read we are making you mad, I would really like to make things just work.
I've tried your core and indeed it fails to locate the compiler. I then added this to platform.keys.rewrite.txt
```
old.5.compiler.path={runtime.ide.path}/hardware/tools/{build.gcc_ver}/bin/
new.5.compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/
```
I can compile with every board you supply (but it breaks `maple mini generic - experimental` because gcc version selection is ignored)

Will adding the above two lines help?
What about a "Add .ZIP hardware" like the already available "add .zip library" ?

ffissore, 2015-04-09:
Ah and yes please, as @MauroMombelli said, the developers list is the best place for looking for improvements

Isaac96, 2015-04-09:
@ffissore 
This fix works great! Thanks. I got rid of the gcc 4.9 option, and will post boards.txt on my repo
 https://github.com/Isaac96/Arduino-stuff

rogerclarkmelbourne, 2015-04-09:
@MauroMombelli and @ffissore 

I will try the fix you suggest.

Also perhaps I need to make a new repo for the stable version of this project and submit a package json file to you guys.

However I was worried that people would think that if they download the STM32F1 board from the boards manager, that it would work 100% compatible to AVR - and as we know, the ARM32 devices are different from the AVR boards and some simple things don't work the same.

MauroMombelli, 2015-04-09:
@rogerclarkmelbourne I'm not a dev into arduino, but i code with STM32f3 so i know there are difference. But there are even for SAM/ARM and galileo (x86!), so i'm quite sure the core has some degree of freedom. Probably even 100%, is up to you to choose what take and what remove.

ps. interesting repo, good work!

rogerclarkmelbourne, 2015-04-10:
@MauroMombelli 

The STM32F3 was a contribution from another user.
I have a STM32F373 but I have not had time to test the STM32F3 folder with my board yet -
there is so much happening at the moment its hard to keep up ;-)

rogerclarkmelbourne, 2015-04-10:
@ffissore 

Re: 
>I can compile with every board you supply (but it breaks maple mini generic - experimental because >gcc version selection is ignored)

I have now removed 2 of the menus, the GCC menu did not work - I think it only worked for Alexey (@hiddenpilot on the Arduino forum), and the "debug" menu he added was the same

While I was testing I also found a bug in the STM32F103C8 menu options  - as it had a wrong linker address, and was linking to the wrong memory location :-(  
So I have fixed this

I also changed name in platform.txt to STM32 Boards.

I will look at the platform re-write .txt file now, but I will need to find where the documentation is for this ???

I guess it was on the developers mailing list ??

I have posted, but using my old GMail account I think.  I will need to add my main email address to the developers mailing list.

Re: Zip file of repo / package.json

I made one of these 1 week ago, but have been too busy to do very much testing.

The problem is that the STM32F103 compatibility with AVR and Due is good, but the STM32F3 and STM32F4 folders are contributions of old code and need a lot of work to update them 

Perhaps I should make a zip package of just the STM32F1 folder, and submit a package.json file to you ?

Thanks

Roger

ffissore, 2015-04-10:
Although we haven't tested it yet, preparing a package_index.json core that depends on the SAME tools used by SAM core should make boards manager installation flawless: if a user has already installed SAM, only STM core will be installed; if a user has NOT yet installed SAM, both STM cores and arm gcc and bossac will be installed.

Yes a zip file is fine, however, keep in mind that we still have no support for such way of adding cores. I suggest you just prepare a package_index.json file, with a link to a tag.gz of your core only. (github released are great for that, especially if you link to tags, see for example how we link adafruit libraries in http://downloads.arduino.cc/libraries/library_index.json)

I'll keep you posted about the additional needs we may have for making the thing work

rogerclarkmelbourne, 2015-04-10:
@ffissore 

No worries ;-)

I made a package file last week.

I will see if I can make a github repo that just contains the STM32F103 boards / cores etc - which are the most stable ones, and perhaps the github zip will be compatible as a download link from the package.json file

This was my experimental package data. (from localhost), so I will make a new repo and retest

```
{
      "name": "RogerClark",
      "maintainer": "Roger Clark",
      "websiteURL": "http://githum.com/rogerclarkmelbourne/arduino-stm32/wiki",
      "email": "roger@rogerclark.net",
      "platforms": [
         {
          "name": "STM32 Boards (32-bits ARM Cortex-M3)",
          "architecture": "STM32",
          "version": "0.1.0",
          "category": "STM32",
          "url": "http://localhost/arduino/STM32F103_0.1.0.zip",
          "archiveFileName": "STM32F103_0.1.0.zip",
          "checksum": "SHA-256:e7e20d3f9e5e85b1e23319b3ecf30325e11c1ce2cd9d0f5b428f98900fb4b3af",
          "size": "19605399",
          "boards": [
            {"name": "STM32F103"}
          ],
          "toolsDependencies": [
            {
              "packager": "arduino",
              "name": "arm-none-eabi-gcc",
              "version": "4.8.3-2014q1"
            }
          ]
        }
      ],
      "tools": [
		{
          "name": "arm-none-eabi-gcc",
          "version": "4.8.3-2014q1",
          "systems": [
            {
              "host": "i686-mingw32",
              "archiveFileName": "gcc-arm-none-eabi-4.8.3-2014q1-windows.tar.gz",
              "url": "http://arduino.cc/download.php?f=/gcc-arm-none-eabi-4.8.3-2014q1-windows.tar.gz",
              "checksum": "SHA-256:fd8c111c861144f932728e00abd3f7d1107e186eb9cd6083a54c7236ea78b7c2",
              "size": "84537449"
            },
            {
              "host": "x86_64-apple-darwin",
              "url": "http://arduino.cc/download.php?f=/gcc-arm-none-eabi-4.8.3-2014q1-mac.tar.gz",
              "archiveFileName": "gcc-arm-none-eabi-4.8.3-2014q1-mac.tar.gz",
              "checksum": "SHA-256:3598acf21600f17a8e4a4e8e193dc422b894dc09384759b270b2ece5facb59c2",
              "size": "52518522"
            },
            {
              "host": "x86_64-pc-linux-gnu",
              "url": "http://arduino.cc/download.php?f=/gcc-arm-none-eabi-4.8.3-2014q1-linux64.tar.gz",
              "archiveFileName": "gcc-arm-none-eabi-4.8.3-2014q1-linux64.tar.gz",
              "checksum": "SHA-256:d23f6626148396d6ec42a5b4d928955a703e0757829195fa71a939e5b86eecf6",
              "size": "51395093"
            },
            {
              "host": "i686-pc-linux-gnu",
              "url": "http://arduino.cc/download.php?f=/gcc-arm-none-eabi-4.8.3-2014q1-linux32.tar.gz",
              "archiveFileName": "gcc-arm-none-eabi-4.8.3-2014q1-linux32.tar.gz",
              "checksum": "SHA-256:ba1994235f69c526c564f65343f22ddbc9822b2ea8c5ee07dd79d89f6ace2498",
              "size": "51029223"
            }
          ]
        }
          ]
        }
```

rogerclarkmelbourne, 2015-04-10:
@ffissore 

If you want a STM32 board, I can post you a Maple mini clone (STM32F103CB), they are very good boards. (they are only $5 and are now all / only clones, because LeafLabs announced end of life their original version last week)

BTW.
I do not have any connection with STM or Leaflabs and don't sell boards etc (I'm just a freelance developer)

email me you postal address to

roger at rogerclark.net  and I will put one in the post to you tomorrow if you want

ffissore, 2015-04-10:
Thank you Roger, very kind. I guess shipping will cost more than the board itself, not to mention the usual carelessness of Italian postal services. We'll buy one on our own. Thank you :)

json file looks good. I hope I'll soon have feedback about the tools issue

rogerclarkmelbourne, 2015-04-10:
Hi Federico

OK now worries about the board. I think can probably send it for slightly less than the cost of the board, as long as I make it a letter instead of a parcel
But I agree, overall its easy and very cheap to but on eBay (not sure of prices to Italy, but perhaps $5 or $6 USD)
Very cheap for 72Mhz Arm board ;-)

rogerclarkmelbourne, 2015-04-10:
Federico

Re: Tools

My repo has a Windows driver installer as well as uploaders etc

I presume uploaders need to be packaged as separate tools ? i.e a bit like AVR Dude

However I'm not sure about the Windows driver installer bat and exe file.

The repo doesn't actually have drivers, it just has a program to register the USB VID/PID numbers against the correct generic Windows drivers. The Serial driver and also the DFU driver.
This is because of the changes to Windows 7 and WIndows 8 about microsoft certified drivers - which are not possible on a community project like the STM32 because there is no money to pay Microsoft

I could package the driver installer bat and exe as tools, but I'm not sure how the user will be able to find them in the %appdata% folder

Is there any way to run the .bat file after installation of the boards package ?

ffissore, 2015-04-10:
yes, there is. each tool can bundle a `post_install.bat`, `install_script.sh` or `post_install.sh`. However we haven't yet tried that with windows drivers

ffissore, 2015-04-10:
warning: tools support a post install script, cores do not (yet)

rogerclarkmelbourne, 2015-04-10:
Federico
OK. That will be interesting

Are there docs for this somewhere ? I can't see an example of that feature in your latest package_index.json file 

Thanks

ffissore, 2015-04-10:
These scripts are not in package_index.json. The IDE looks for them once it has unpacked a tools. Intel uses them in their core2-32-poky-linux tool for linux. So you need to pack them along with the tool itself and they will be executed.

ffissore, 2015-04-10:
FYI https://github.com/arduino/Arduino/issues/2956

rogerclarkmelbourne, 2015-04-10:
Thanks I'll look

rogerclarkmelbourne, 2015-04-12:
I have added platform.rewrite.keys.txt to the STM32F1 folder and it now seems to work with 1.6.3

However, I had to change platform.txt because there appears to be a bug in the re-write code as it doesn't seem to handle substitution

If you look at

https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/4a3da022b8fcfe3daf65a46d77f152c0666a6539#diff-5ae7354fa4b9d8739cc748960851d64b

You can see that I had to remove the substitution in compiler.path 

I suspect that the substitution for build.gcc_ver=gcc-arm-none-eabi-4.8.3-2014q1 is being processed before the rewrite.keys, hence it didn't work for

compiler.path={runtime.ide.path}/hardware/tools/{build.gcc_ver}/bin/

Its probably a waste of time trying to get this to work, as it may break other things.

But I will post it as an issue, just for the record ;-)

rogerclarkmelbourne, 2015-04-12:
PS.
If I make a Boards Manager package, is there a process to submit it to Arduino.cc ??

I notice that you currently do not have any third party boards listed, and Adafruit are  now serving their own replacement package_index.json file using a hack involving the use of a local proxy.

Its pointless me making a boards manager package, and having a modified version of the Adafruit proxy server, just for one board.

Isaac96, 2015-04-12:
@rogerclarkmelbourne 
Please add STM32 to boards manager! It will make the project a lot more popular and easier to use (It's already not very hard).
(I can't log into my forum account, so I probably won't be posting much).

rogerclarkmelbourne, 2015-04-12:
@Isaac96 

There is currently no way to add packages to the boards manager.

Its all controlled by Arduino.cc, and currently the only boards I see listed are AVR, SAM and Intel Galileo 

None of the Adafruit boards are in the list yet.

So if Adafruit can't get their boards added, then there is very little chance of a non-commercial board getting added.

The only way to add boards is via a hack, either the Adafruit proxy hack, or there is a way to change the download location by changing a file that Java uses.

Until the IDE dev team have a method to either manually add repositories to the boards manager, or accept submissions of package files, no other boards are going to appear in the list.

Isaac96, 2015-04-12:
@rogerclarkmelbourne 
The README for the repo needs updating.
-Isaac

rogerclarkmelbourne, 2015-04-12:
Hi @Isaac96 

Thanks. I'd updated the Wiki pages in several places but forgot to update the readme

I've just removed the stuff I added about 1.6.2, as the details of how to install on 1.6.2 (and newer) i.e using the boards manager to install the SAM based boards is now in the wiki.

I will close this issue, as the repo is now compatible with 1.6.2 +

BTW. I noticed that Federico has closed the other issues regarding installing via the boards manager.

At the moment its unclear when, or if, the Dev team will allow third party boards to be installed via the boards manager.

closed 2015-04-12 by rogerclarkmelbourne

ffissore, 2015-04-13:
platform.keys.rewrite is not expected to be provided by a core, it's rather a hack the IDE uses to make a core compatible without waiting for upstream to port it. that's probably why it's not working as you expect.

as for adding custom boards, the how and when is being discussed in this very moment

ffissore, 2015-04-13:
And it's being tracked by https://github.com/arduino/Arduino/issues/2949