rogerclarkmelbourne/Arduino_STM32 #40: Linux Installation

redafenj, 2015-04-06:
Hello,

I tried to install your arduino STM32 IDE for my Olimex board, but I do not succeed to get the STM32 Board Selection in the Arduino IDE. This is may be I made a mistake where I copy the Arduino_STM32.

I tried to copy it under ~/Arduino/hardware/Arduino_STM3 it is not working
I tried to copy where I download the linux install link : 
~/dvpt_tools/arduino/arduino_stm32/arduino-1.6

~/dvpt_tools/arduino/arduino_stm32/arduino-1.6.1$ ls
arduino  Arduino_STM32  examples  hardware  lib  libraries  reference  revisions.txt  tools
But still it does not work.

Could you please help to figure out how I can my Linux STM32 Arduino IDE working ?

Thanks

rogerclarkmelbourne, 2015-04-08:
Sorry, I can't personally support individual installation problems 

Please post to the Arduino forum

http://forum.arduino.cc/index.php?topic=265904.new#new

There are several people on the forum who use Linux and may be able to assist you

rogerclarkmelbourne, 2015-04-12:
Repo has been updated to work with IDE 1.6.2 and newer, the wiki has also been updated.

Unfortunately I don't have Linux so I'm unable to test this at the moment, but please try the latest copy of the repo

closed 2015-04-12 by rogerclarkmelbourne