rogerclarkmelbourne/Arduino_STM32 #43: stm32f4 and i2s to get codec working.

gpud, 2015-04-29:
Hi and thanks for your port of Arduino to STM32f4. I've got it working and was wondering if i2s is available to get the stm32f4 discovery's audio codec working?
Thanks again.
Steve.S

madias123, 2015-04-29:
Please post you wishes also in the new forum!
Maybe some of them has worked with I2s (I don't think so!)
http://www.stm32duino.com/index.php

rogerclarkmelbourne, 2015-04-29:
@gpud

Not sure about Wire on the F4, as I've not tested it.

However on the F103 its a software implementation (bit banged) so I expect its the same on the F4

There were some minor bug fixes to the F103 version, so even if the F4 I2C works, I'd recommend you copy the Wire folder from STM32F1/libraries and give that one a try.

I can't recall all the fixes, but the important one was that the example sketch "i2C Scanner" didnt work, until the fix.

Re: Codec
Sorry no idea. Probably not. The initial port from F103 to F4 was done by the AeroQuad team for use on quadcopters, so I doubt they'd have need of an audio codec ;-)

If you care to join the forum (if youve not joined already) I added a F4 boards  see
http://www.stm32duino.com/viewforum.php?f=39

rogerclarkmelbourne, 2015-05-03:
There isnt and I2S library for any of the boards (not even STM32F1)

This will need to be written by someone. I personally have more urgent things on the project that need my attention.

I recommend you read the programming reference

http://www.st.com/web/en/resource/technical/document/reference_manual/CD00171190.pdf

and probably duplicate and modify the SPI lib, as I2S on STM32 seems to be an additional function within the SPI section of the chip

rogerclarkmelbourne, 2016-11-10:
STM are now writing their own Core for the Nucleo boards, which will include a Nucleo F4, the best option is probably to wait for them to release a the code for the Nucleo F4 and then make a modified version of that to suit your boards

"Watch" the stm32duino github account

closed 2016-11-10 by rogerclarkmelbourne