rogerclarkmelbourne/Arduino_STM32 #45: linux tools missing

RickKimball, 2015-04-29:
missing binaries for stlink (both 32 and 64)
missing binaries for stm32flash ( both 32 and 64)
missing script serial_upload
missing script stlink_upload

rogerclarkmelbourne, 2015-04-29:
Rick

Can you do me a favour

Go into 

https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/tools/src

And Make the binaries for your platform and somehow send them to me ;-)

RickKimball, 2015-04-29:
https://github.com/RickKimball/Arduino_STM32/tree/master/tools/linux/stm32flash/i686

rogerclarkmelbourne, 2015-04-29:
Rick

Is that binary x86 or x64 ?

I need to do 2 separate folders for the different arcs :-(

rogerclarkmelbourne, 2015-04-29:
PS. We don't have a DFU binary either

There are instuctions in the Wiki on how to build

But if we can just host x84 and 64 bit versions it would make it easier for all linux users

PS.  I'm not a Linux expert
So just to confrim if we build both x84 and 64 bit versions will they work on all distros ?

RickKimball, 2015-04-29:
it is a 32 bit .. but I'm running the i686 .. probably more universal if it was i386

rogerclarkmelbourne, 2015-04-29:
OK..

So i686 is 64 bit ?

Sorry for my ignorance but does linux 64 (i686)  bit run 32 bit binaries ?

My best bet is to create a 32 bit linux vm on my Windows machine or my Mac

RickKimball, 2015-04-30:
i686 is a 586 ( 32 bit )
x86_64 is 64 bit

rogerclarkmelbourne, 2015-04-30:
Rick

OK 

Thanks.

I'll put your version into the Linux folder, and wait for the x86_64 people to complain it doesn't work ;-)

rogerclarkmelbourne, 2015-05-03:
Tools have been added

I compiled 32 bit versions of stm32flash and st-flash and have tested on 32 bit and 64 bit linux and they seem to work Ok for me when tested using the "STM32f103c8 to flash (no bootloader)" board type

I will update boards.txt for the other generic boards to make these options available

RickKimball, 2015-05-05:
Seems fixed now

closed 2015-05-05 by RickKimball