rogerclarkmelbourne/Arduino_STM32 #46: missing stlink upload option for Generic STM32F103C

RickKimball, 2015-04-29:
http://www.stm32duino.com/viewtopic.php?f=35&t=27&p=130#p119 This is one seems to work for the stm32f103c8 .. still need one for the CB version

rogerclarkmelbourne, 2015-04-30:
Rick

I've spent some time trying to tidy the boards.txt and platform.txt files up.

Here is the cleanest version of the STM32F103Z series that I've been able to get to compile and upload

```
##################################################################################################################
genericSTM32F103Z.name=Generic STM32F103Z Series

genericSTM32F103Z.build.error_led_port=GPIOG
genericSTM32F103Z.build.error_led_pin=15
genericSTM32F103Z.build.variant=generic_stm32f103zxx
genericSTM32F103Z.build.common_flags=-mthumb  -march=armv7-m -D__STM32F1__ 

genericSTM32F103Z.build.mcu=cortex-m3
genericSTM32F103Z.build.f_cpu=72000000L
genericSTM32F103Z.build.board=GENERIC_STM32F103Z
genericSTM32F103Z.build.core=maple
genericSTM32F103Z.build.vect=VECT_TAB_ADDR=0x8000000

genericSTM32F103Z.menu.device_variant.STM32F103ZC=STM32F103ZC
genericSTM32F103Z.menu.device_variant.STM32F103ZC.build.cpu_flags=-DMCU_STM32F103ZC  
genericSTM32F103Z.menu.device_variant.STM32F103ZC.upload.maximum_size=262144
genericSTM32F103Z.menu.device_variant.STM32F103ZC.upload.ram.maximum_size=492152
genericSTM32F103Z.menu.device_variant.STM32F103ZC.upload.flash.maximum_size=262144
genericSTM32F103Z.menu.device_variant.STM32F103ZC.build.ldscript=ld/stm32f103zc.ld

genericSTM32F103Z.menu.device_variant.STM32F103ZD=STM32F103ZD
genericSTM32F103Z.menu.device_variant.STM32F103ZD.build.cpu_flags=-DMCU_STM32F103ZD  
genericSTM32F103Z.menu.device_variant.STM32F103ZD.upload.maximum_size=393216
genericSTM32F103Z.menu.device_variant.STM32F103ZD.upload.ram.maximum_size=65536
genericSTM32F103Z.menu.device_variant.STM32F103ZD.upload.flash.maximum_size=393216
genericSTM32F103Z.menu.device_variant.STM32F103ZD.build.ldscript=ld/stm32f103zd.ld

genericSTM32F103Z.menu.device_variant.STM32F103ZE=STM32F103ZE
genericSTM32F103Z.menu.device_variant.STM32F103ZE.build.cpu_flags=-DMCU_STM32F103ZE
genericSTM32F103Z.menu.device_variant.STM32F103ZE.upload.maximum_size=524288
genericSTM32F103Z.menu.device_variant.STM32F103ZE.upload.ram.maximum_size=65536
genericSTM32F103Z.menu.device_variant.STM32F103ZE.upload.flash.maximum_size=524288
genericSTM32F103Z.menu.device_variant.STM32F103ZE.build.ldscript=ld/stm32f103ze.ld

genericSTM32F103Z.menu.upload_method.serialMethod=Serial
genericSTM32F103Z.menu.upload_method.serialMethod.upload.protocol=maple_serial
genericSTM32F103Z.menu.upload_method.serialMethod.upload.tool=serial_upload

genericSTM32F103Z.menu.upload_method.STLinkMethod=STLink
genericSTM32F103Z.menu.upload_method.STLinkMethod.upload.protocol=STLink
genericSTM32F103Z.menu.upload_method.STLinkMethod.upload.tool=stlink_upload
genericSTM32F103Z.menu.upload_method.STLinkMethod.build.debug_flags=-DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1
```

I've spotted several mistakes in platform.txt as well, but none of them actually stopped things working ;-)

```
# For more info:
# https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification

name=STM32 Boards
version=0.1.0

# compiler variables
# ----------------------
compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/
compiler.c.cmd=arm-none-eabi-gcc
compiler.c.flags=-c -g -Os -w -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -DBOARD_{build.variant} -D{build.vect} -DERROR_LED_PORT={build.error_led_port} -DERROR_LED_PIN={build.error_led_pin}
compiler.c.elf.cmd=arm-none-eabi-g++
compiler.c.elf.flags=-Os -Wl,--gc-sections
compiler.S.cmd=arm-none-eabi-gcc
compiler.S.flags=-c -g -x assembler-with-cpp -MMD
[... 120 lines cut ...]
tools.stlink_upload.path.linux={runtime.hardware.path}/tools/linux
```

I'm not sure what params.verbose=-d does, but the IDE wants it.
```
tools.stlink_upload.upload.params.verbose=-d
```

finally the whole lot of put together to make one big string  from the stuff built above
i.e this is the command that the IDE executes for the upload
```
tools.stlink_upload.upload.pattern="{path}/{cmd}" {serial.port.file} {upload.altID} {upload.usbID} "{build.path}/{build.project_name}.bin"
```

Note we have redundant items for USB ID etc, which are only applicable to the Maple uploader

This is my fault, I just copied the maple uploader code when I created the other ones, I should have worked out that the USB ID and ALT ID were not needed for serial or STLink

And even the Serial port is redundant on STLink I think

So we could clean it all up, but you'd need to modify the serial_upload and stlink_upload (bash and windows scripts) to get them to match the number of arguements

I didnt want to push my changes, as I've been doing some other things as well, and also I didnt want to break anything, ;-)

Anyway, I hope that helps,... Or I may be telling you stuff you know already ;-)

RickKimball, 2015-04-30:
I'm all for cleaning it up. However, right now I only have a few smt32f103c8 and a few stm32f0 discovery/nucleo boards.  I can't really test any of that out except to do inspection.  And to be honest now that my c8 board works .. heh .. well .. i'm happy.

I think stability right now is a good thing.  At least as a bunch of new people check it out. But beware you don't have long before making any changes will be bring groans of change.

rogerclarkmelbourne, 2015-04-30:
Rick

Part of the cleaning is to be able to have the flags for the debug stuff needed for stlink, to be separated from the build_extra_flags

So I definitely need to do that

But perhaps I will limit the change to just adding the new debug_flags parameter

Its a delicate balancing act, as half the time people complain that some things are overly verbose, and the other half would rather I just add something rather than doing a clean up 

Anyway. I will sort something out ;-)

RickKimball, 2015-05-01:
I'm not sure if you had a chance to checkout the commits to the boards.txt I made on my fork.  I have it setup to properly compile for stlink and stm32flash.  In my fork I can compile and upload using both stlink and stm32flash right from the arduino ide now on linux.  Personally I think leaving the debug pins on by default is a useful thing because you can always use it later to debug even if your normal mode is to use a serial loader.  When you are used to less than 28 pins of useable pinage .. leaving 2 alone on  board with > 30 isn't that big a deal

rogerclarkmelbourne, 2015-05-01:
Rick,

Sorry, I didn't get chance to look at your repo.

Re: leaving that define on all time time. I'll need to check if it affects the Maple mini guys, i.e do they loose GPIO pins ?

I didnt remove that option, it was possibly something Bob Cousins did when he did his initial port of libmaple , which I then used for the repo

PS. Bob has moved on to new things and is no longer participating, mainly because of a difference of opinion with one specific respondent on the Arduino forum.
(its a shame but thats the way the cookie crumbles)

rogerclarkmelbourne, 2015-05-03:
I have added the binaries so that Linux can upload via STLink or ST Flash, however I still need to update the Generic boards in boards.txt to facilitate using the binaries

I was waiting for @ahull to let me know if he thinks he needs upload to 0x8005000 even though he's not using the bootloader, as these options IMHO are pointless, and also clutter and confuse the menu structure

I will update the comment when I get the all clear from @ahull

RickKimball, 2015-05-05:
Seems to be fixed now

closed 2015-05-05 by RickKimball