rogerclarkmelbourne/Arduino_STM32 #47: resulting binaries (.elf files) have extra stuff

RickKimball, 2015-04-30:
The platform.txt recipe for doing linking uses these arguments to link in the core.a file:

-Wl,--whole-archive "{build.path}/{archive_file}" -Wl,--no-whole-archive

at line: https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/STM32F1/platform.txt#L82

Is there a reason we do this?  It sucks in the entire core.a file instead of just searching and including the routines it actually uses. This results in a  13k binary for a bare minimum sketch.  

If you drop the ld directives and just use:
"{build.path}/{archive_file}"

Then you end up with a more reasonable 3k .elf

Is there a reason to force inclusion of all files in the core.a?

rogerclarkmelbourne, 2015-04-30:
If we don't compile in the whole archive, some things stop working,

Serveral other people have tried and unfortunately failed to figure out why.

If you look back in the history for platform.txt you will see that Whole archive has been removed but had to be reinstated

I think the last reason it was reinstated was because attachTimer callbacks didn't seem to work without it

People have speculated that its something to do with weak references in the code. A lot of stuff is defined as week weak in the lower levels of the code.

Unfortunately the problem with this issue is that changing it , potentially changes everything and people get quite upset when stuff that was working, suddenly stops working.

Hence considering it only saves a few kb by removing it, the safer option is to leave it in, until someone has time to do complete testing of everything.

RickKimball, 2015-04-30:
Do you have some specific code that doesn't work? Everything I tried worked. * mostly serial stuff *  I'd be glad to figure it out.

RickKimball, 2015-05-07:
So can we think about doing this? With the recent pin map changes you can reduce the size significantly:

    Sketch uses 3,740 bytes (5%) of program storage space. Maximum is 65,536 bytes.
    Global variables use 1,360 bytes of dynamic memory.

rogerclarkmelbourne, 2015-05-07:
Rick,

Have you tested attachInterrup() your latest change, from what I recall thats the thing which broken when the whole archive flag was removed,  if you look in the commit history I probably wrote the reason why I had to last re-instate the flag

RickKimball, 2015-05-07:
I looked at the history on platforn.txt and didn't see any changes or comments about the change

rogerclarkmelbourne, 2015-05-07:
https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/88d8e2e7be2b4f9bf289591ccaa7b048ca45347b

RickKimball, 2015-05-26:
http://www.stm32duino.com/viewtopic.php?f=16&t=148#p1343 more info there about the weak references causing problems.

rogerclarkmelbourne, 2015-05-27:
Rick,

I was hoping some other uses would take a look at the branch https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/fix_weak_ref_timers and test it.

i.e it works for me, and its your fix, so it should work for you, but we are a limited subset.

However if in a few days no one has bothered.

I will just release it anyway, as I think we are both confident its no worse than we had before, and is hopefully considerably better

rogerclarkmelbourne, 2015-05-30:
This will be actioned, its pending in a branch for testing.

I will close this issue when the branch has been merged into the master

rogerclarkmelbourne, 2015-05-31:
fixed as part of commit 
https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/9411f039887cae692511fe483b2ddb79a00990f3

closed 2015-05-31 by rogerclarkmelbourne