rogerclarkmelbourne/Arduino_STM32 #54: __always_inline macro conflicts with gcc definition

RickKimball, 2015-05-07:
the macro '__always_inline' is defined in both the libmaple and also in the gcc headers. The problem is that they don't do the same thing. gcc doesn't include the inline keyword it is just the __attribute__ part.

Two fixes. 

o change all the code in libmaple to add a an inline keyword to precede any __always_inline macro usage
    or
o rename the macro so people don't aren't confused if they are used to using the one from gcc

I'd opt for the second solution as the libmaple is different than the gcc

rogerclarkmelbourne, 2015-05-23:
Leaflabs seem to changed all the __always_inline to inline in their repo in the past few months.

I as going to attempt the same method to fix

rogerclarkmelbourne, 2015-05-27:
Rick,

I think I will eventually go with your renaming idea for this, but I have a number of other more pressing things to get fixed, e.g. stuff that is broken, rather than this, which is not critical at the moment (though I appreciate it does need to be done, as I hate warnings as much as the next person)

RickKimball, 2015-05-28:
I've actually gone through and just modified the code and made the mapleleaf __always_inline track the way the gcc one is done.

I've done it  in my branch .. however if you like the idea I can do it for your latest master code

https://github.com/RickKimball/stm32duino_rk/commit/7a5eb47b1eca6112c461e11bd57e9ec2b7aad09b

and later I found I missed one:
https://github.com/RickKimball/stm32duino_rk/commit/475c0bc2aec13183947df1571c6a0c4680db4498

rogerclarkmelbourne, 2015-05-28:
Hi Rick, 

It looks good, but I'd like to release the --whole-archive version first rather than have a whole lot of possibly conflicting stuff released in one go.

I know that there is theoretically no relation between these two changes, but I'm just cautious of rolling out changes across the whole of the core, as I've been burnt 4 or 5 times before with this sort of stuff, so would prefer to release incrementally, even if it means delaying a week or two for the dust to settle between each change.

RickKimball, 2015-05-28:
Here is a patch I did today .. if you make more changes to the master .. they may or may not work

https://gist.github.com/RickKimball/7f2ddbac870d8faaaeb0

rogerclarkmelbourne, 2015-05-28:
Thanks  rick

rogerclarkmelbourne, 2015-05-30:
This will be actioned when I get chance

rogerclarkmelbourne, 2015-11-20:
Its still on my todo list.

I noticed you have changed this in your repo, I will try to find time to put the same change into the development branch.

RickKimball, 2016-05-27:
It seems you committed this on May 3rd. Thanks!

closed 2016-05-27 by RickKimball

rogerclarkmelbourne, 2016-05-28:
Yes.
Sorry. I forgot to mention I'd done it ;-)