rogerclarkmelbourne/Arduino_STM32 #56: Need to manually take newer commits from LeafLabs/libmaple into this repo

rogerclarkmelbourne, 2015-05-07:
@RickKimball

I've noticed that the original leaflabs libmaple repo has some fixes that also need to be manually applied to this repo.

https://github.com/leaflabs/libmaple/commits/master

Long story, but the original code for this repo was copied by BobC from https://github.com/leaflabs/libmaple/commit/f4d4fe2e55281cc3190b006bead7b0050687f008

He made some changes to get it to work with the 1.5.x IDE

I copied his repo when he stopped working on it, and made loads of changes to bring it roughly up to Arduino API 1.0

Which basically means, we don't have the fixes that have been added to the leaflabs repo since 

https://github.com/leaflabs/libmaple/commit/f4d4fe2e55281cc3190b006bead7b0050687f008

So there are possible fixes for some of the issues you raised
e.g.

https://github.com/leaflabs/libmaple/commit/f105675de87d19c8d627ccf87e3366d7f6ebc3aa

However I suspect as the code has changed quite a lot, it will take some detective work to work out the current location of some of the code in question

madias123, 2015-05-18:
I don't think that there are too much fixes left, maybe this one:
USART hangs on read overrun #107: https://github.com/leaflabs/libmaple/issues/107 from ctz.

rogerclarkmelbourne, 2015-05-18:
Hi Matthias

I tried manually applying all the fixes into a branch of the repo, 

https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/apply-libmaple-updates

but somehow it completely wrecked it and it didnt work any more ;-( 

I think the problem is possibly the replacement of the "__always_inline" with "inline" which happened in about 20 places, with the purpose of getting rid of the compiler warnings

I think what I'll need to do is to go back to the first checkin of that branch and test build each of the commits one by one, rather than what I actually did, which was to do all the changes one by one, but only rebuilding to test at the end.

I think the change in #107 is likely to be low risk and didn't cause the problems that I have with that branch

rogerclarkmelbourne, 2015-05-31:
I have manually applied the fixes to usart_private.h and dma_private and committed 

See

912098d5b84a0c73d7743c815c6c469a134af99b

I also put one other small change into this commit, to prevent the compiler from warning about unused args in USBSerial:begin()

I will implement the fixes for __always_inline, in a separate commit, as there is a separate issue for this

closed 2015-05-31 by rogerclarkmelbourne