rogerclarkmelbourne/Arduino_STM32 #72: White space is all over the place

RickKimball, 2015-05-29:
I'm looking at the new commits going into the tree and everyone is using different white space settings. Could you document what white space settings should be used?

And then not accept push requests that aren't whitespace clean?

Serasidis, 2015-05-29:
Sorry but Github is a new tool for me. I am still learning how it works.
Could you please help me on this "white space" issue ?

RickKimball, 2015-05-29:
Different code editor methods to indent. Some use spaces, some use tabs and some tab width.  If we all use different tab widths and spacing for indenting then the code looks very strange when people make changes. 

If we document how white space (\r\n, \n, \b , \t ) are used then it will be easier to have code that looks the same for everyone.

Serasidis, 2015-05-29:
I use Notepad++ as it is my favorite text editor. 
I setup the Notepad to use 2 white spaces for every depth change (for {...}) 
Moreover, I choose to convert any Tab into 2 white spaces.

[image]

RickKimball, 2015-05-29:
Most of the maple code I've looked at is setup has tabs setup to 4 or the code is using 4 spaces

rogerclarkmelbourne, 2015-05-29:
Vasillis

I use Notepad++ as well.

I noticed that lots of code has different spacings, but as AFIK, white space doesn't effect the function of a C program. I'm not sure how this is technical issue. (unlike some other languages)

Please correct me if I'm wrong

RickKimball, 2015-05-29:
It isn't a technical issue. .. I guess it is fine.

RickKimball, 2015-05-30:
I guess anything goes.

victorpv, 2015-05-31:
I'm thinking this kind of stuff should have a section in the forums called:
"Coding standards"
Some guidelines, if you don't follow them nothing breaks, if you follow them is easier to read, so I think is a subject for the forum.

rogerclarkmelbourne, 2015-05-31:
This is loosely coding standards, but wasn't rick's issue that the file has strange tab settings.
I suspect that the tab settings may not even have been defined by Vasillis, but as the library was ported I suspect that the tab settings may well have come from the original author of the library.

Its asking a bit much of anyone porting a library, that they reformat all the code.

In terms of coding standards, I'd be more inclined to be concerned about things like this

```
  while (len)
  {
	uint16_t addr = pgm_read_word(buf++); len--;
    uint16_t n = pgm_read_word(buf++); len--;
```

Which appear to be in the original library

https://github.com/maniacbug/VS1053/blob/master/VS1053.cpp

(a)  Why are addr and n both defined inside a while loop ???
(b)  Why is len--; not on its own line ?
(c) Why is len being decremented twice when the second line doesn't use len e.g.

```
    uint16_t n = pgm_read_word(buf++); 
```
doesn't use len, do why bother decrementing it after setting up addr

However I think our time is better spent fixing issues and porting other libraries etc, rather than nit picking though existing libraries that work OK.

After all, the whole Arduino and Open Source ethos is that anyone can write a library and publish it.

If code formatting is really such an issue, please feel to run lint on the code, however also please be sure to test your reformatted code on the hardware it supports, just on the offchance that somehow the reformatting went wrong.

closed 2015-06-02 by rogerclarkmelbourne