rogerclarkmelbourne/Arduino_STM32 #92: Possibly dated driver instructions in wiki

realblades, 2015-07-15:
https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Maple-and-Maple-mini

This page says driver signing needs to be disabled, but I don't believe that's necessary anymore.

https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Maple-drivers

This page mentions the dwi-simple that I think worked fine without the setting. Using Zadig is also a way to get around the limitation for certain driver libraries.

http://zadig.akeo.ie/

rogerclarkmelbourne, 2015-07-15:
Sorry

The docs are out of date. We now use libwdi which seems to work fine (I've not had anyone complain the driver install doesnt work)

I will update the wiki

rogerclarkmelbourne, 2015-07-15:
Wiki has been updated

closed 2015-07-15 by rogerclarkmelbourne