rogerclarkmelbourne/Arduino_STM32 #93: Arduino 1.6.5 IDE and Board manager messages

slavkoja, 2015-07-23:
Hi,

i tried to follow installation instructions for Linux, i installed the `Arduino_STM32` under the `~/Arduino/hardware` directory and it seems to work (i don't tried the upload yet), but anytime when i open the Board manager, i get these messages in Arduino's debug window:
```
Could not find boards.txt in /home/XXXX/Arduino/hardware/Arduino_STM32/tools. Is it pre-1.5?
Could not find boards.txt in /home/XXXX/Arduino/hardware/Arduino_STM32/drivers. Is it pre-1.5?
Warning: platform.txt from core 'STM32 Boards (STM32duino.com)' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/, automatically converted to compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/. Consider upgrading this core.
Board Arduino_STM32:STM32F3:discovery_f3 doesn't define a 'build.board' preference. Auto-set to: STM32F3_DISCOVERY_F3
Warning: platform.txt from core 'STM32F3 boards' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/, automatically converted to compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/. Consider upgrading this core.
Board Arduino_STM32:STM32F4:discovery_f407 doesn't define a 'build.board' preference. Auto-set to: STM32F4_DISCOVERY_F407
Board Arduino_STM32:STM32F4:stm32f4stamp doesn't define a 'build.board' preference. Auto-set to: STM32F4_STM32F4STAMP
Board Arduino_STM32:STM32F4:netduino2plus doesn't define a 'build.board' preference. Auto-set to: STM32F4_NETDUINO2PLUS
Warning: platform.txt from core 'STM32F4 boards' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/, automatically converted to compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/. Consider upgrading this core.
Could not find boards.txt in /home/XXXX/Arduino/hardware/Arduino_STM32/examples. Is it pre-1.5?
Could not find boards.txt in /home/XXXX/Arduino/hardware/Arduino_STM32/tools. Is it pre-1.5?
Could not find boards.txt in /home/XXXX/Arduino/hardware/Arduino_STM32/drivers. Is it pre-1.5?
Warning: platform.txt from core 'STM32 Boards (STM32duino.com)' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/, automatically converted to compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/. Consider upgrading this core.
Board Arduino_STM32:STM32F3:discovery_f3 doesn't define a 'build.board' preference. Auto-set to: STM32F3_DISCOVERY_F3
Warning: platform.txt from core 'STM32F3 boards' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/, automatically converted to compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/. Consider upgrading this core.
Board Arduino_STM32:STM32F4:discovery_f407 doesn't define a 'build.board' preference. Auto-set to: STM32F4_DISCOVERY_F407
Board Arduino_STM32:STM32F4:stm32f4stamp doesn't define a 'build.board' preference. Auto-set to: STM32F4_STM32F4STAMP
Board Arduino_STM32:STM32F4:netduino2plus doesn't define a 'build.board' preference. Auto-set to: STM32F4_NETDUINO2PLUS
Warning: platform.txt from core 'STM32F4 boards' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/, automatically converted to compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/. Consider upgrading this core.
Could not find boards.txt in /home/XXXX/Arduino/hardware/Arduino_STM32/examples. Is it pre-1.5?
```
I don't understand this in depth, but when i take look, i can see, that the Arduino Duo code was installed under `~/.arduino15/packages/arduino/hardware/sam` and the tools are in the `~/.arduino15/packages/arduino/tools`, then IMO some improvements of the location are needed.

BTW: i replaced my account homedir with `XXXX`.

rogerclarkmelbourne, 2015-07-23:
The reason that these messages appear is because my repo contains folders for things like the bootloader and windows drivers, which are not structly speaking "core" files as defined in the Arduino specification.

However i dont want to remove these folders at the moment, otherwise people would have to download them separately.

One workaround that i have seen used, is to put empty board.txt files in the other folders, as this supresses these messages.

However in the long term I will migrate to a Boards Manager package, in which case I have to completely restructure the whole repo, and in which case i expect i will move the files to a completely new account on github.

In the mean time, please ignore these messages.

closed 2015-07-23 by rogerclarkmelbourne

silverdr, 2017-08-10:
Obviously the best way would be to go with Boards Manager package yet - personally - I'd very much prefer download documented things separately than have to work (even harmless) incompatibilities around.