rogerclarkmelbourne/Arduino_STM32 #95: openocd support?

lloyd-g, 2015-07-29:
Any chance as I don't have a ST-link?
but have RLINK and a DIY JTAG that works with OPENOCD?
I have a board with RLINK built in.

rogerclarkmelbourne, 2015-07-29:
What S/W do you normally use to upload to using RLink?

Also, have you looked at the Black Magic Probe option in the upload menu? 

This uses GDB to do the upload, and I thought that OpenOCD was compatible with GDB, however I've not used Black Magic probe (which is now open source hardware)

I think your best option is to register on stm32duino.com and PM Rick Kimball, as he did all the work on the GDB / Black Magic Probe upload and he can probably tell you how to get it working with OpenOCD

lloyd-g, 2015-07-29:
have used OPENOCD on Ubuntu for RLINK. I had to write a new cfg script. works well with pain STM32 code but everything gets long winded. I what to write quick code with USB serial debugging like the Arduino.
I can use that if the IDE would just output an elf or Hex file?
and then run my bash script.

rogerclarkmelbourne, 2015-07-29:
Yes,

You can just use the binary file.
If yiu turn on verbose in the preferences, you will see where the IDE puts the binary.

However it would be really easy for you to modify one of the existing scripts e.g the stlink upload script to use your hardware instead.

if you look in tools/linux you will see a load of scripts including serial upload and stlink upload

e.g.

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/tools/linux/stlink_upload

see line 31 where it calls st-flash to perform the upload

just edit that line and replace with the command to use your programmer

selecting the stlink upload option gives you USB serial. if you dont want this, use serial upload and modify that script instead

rogerclarkmelbourne, 2015-11-20:
Closed due to lack of response

closed 2015-11-20 by rogerclarkmelbourne