rogerclarkmelbourne/Arduino_STM32 #100: Arduino throws a lot of warnings to stdout (stderr) when using Arduino_STM32

LieBtrau, 2015-08-21:
I'm using Arduino 1.6.5 as a command line compiler for a Nucleo project.  I noticed that quite a lot of warnings are thrown to stdout (or stderr) when running Arduino from a command line window:

Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Could not find boards.txt in /home/ctack/Programs/arduino-1.6.5/hardware/Arduino_STM32/examples. Is it pre-1.5?
Board Arduino_STM32:STM32F3:discovery_f3 doesn't define a 'build.board' preference. Auto-set to: STM32F3_DISCOVERY_F3
Warning: platform.txt from core 'STM32F3 boards' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/, automatically converted to compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/. Consider upgrading this core.
Could not find boards.txt in /home/ctack/Programs/arduino-1.6.5/hardware/Arduino_STM32/drivers. Is it pre-1.5?
Warning: platform.txt from core 'STM32 Boards (STM32duino.com)' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/, automatically converted to compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/. Consider upgrading this core.
Board Arduino_STM32:STM32F4:discovery_f407 doesn't define a 'build.board' preference. Auto-set to: STM32F4_DISCOVERY_F407
Board Arduino_STM32:STM32F4:stm32f4stamp doesn't define a 'build.board' preference. Auto-set to: STM32F4_STM32F4STAMP
Board Arduino_STM32:STM32F4:netduino2plus doesn't define a 'build.board' preference. Auto-set to: STM32F4_NETDUINO2PLUS
Warning: platform.txt from core 'STM32F4 boards' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/, automatically converted to compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/. Consider upgrading this core.
Could not find boards.txt in /home/ctack/Programs/arduino-1.6.5/hardware/Arduino_STM32/tools. Is it pre-1.5?
Preparing boards...
Verifying...

To make things clear: building projects for the Nucleo works just fine with Arduino_STM32 and Arduino, except for the above errors being thrown.

rogerclarkmelbourne, 2015-10-26:
umm

Can you retest
I've added empty boards.txt files to tools, drivers and examples and as far as I can see 

stm32f4stamp.name=STM32F4Stamp F405

now exists

Note the platform.txt deprecated stuff is correct

If I change it, any IDE version before 1.6.2 would cease to work.

However I will poll the users on stm32duino.com and find out if everyone is happy to no longer support older ide versions

LieBtrau, 2015-10-27:
Hello Roger,

The issue is only about the tool generating some warnings.  Things were still working.  Many warnings have been fixed:
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Warning: platform.txt from core 'STM32F3 boards' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/, automatically converted to compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/. Consider upgrading this core.
Could not find boards.txt in /home/ctack/Programs/arduino-1.6.5/hardware/Arduino_STM32/STM32duino-bootloader. Is it pre-1.5?
Warning: platform.txt from core 'STM32 Boards (STM32duino.com)' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/, automatically converted to compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/. Consider upgrading this core.
Warning: platform.txt from core 'GD32 Boards (STM32duino.com)' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/, automatically converted to compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/. Consider upgrading this core.
Warning: platform.txt from core 'STM32F4 boards' contains deprecated compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/, automatically converted to compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/. Consider upgrading this core.
Preparing boards...
Verifying...

You're right about asking users if they still want to see support for older Arduino versions.  Let's hope that they favor breaking compatibility so that you don't have to drag old junk along until eternity.

rogerclarkmelbourne, 2015-10-27:
No worries.

All the votes so far have been to move to only supporting the newer versions of the IDE.

So I should be able to go ahead and change this.

However I will give the poll some more time, as I think only 10 people voted so far

rogerclarkmelbourne, 2016-07-18:
The repo now runs on IDE 1.6.9 with virtually no warnings, and I will soon remove 1.6.5 support so that we do not get any warnings at all

closed 2016-07-18 by rogerclarkmelbourne