rogerclarkmelbourne/Arduino_STM32 #103: Adding support for STM32F401?

draggett, 2015-08-26:
I have installed your STM32 support for the Arduino IDE (1.6.5) and would like to see if I can extend this to support the STM32F401.  I am guessing that this is mostly a matter up updating boards.txt in your STM32F4 folder, and adding  a new section based on one of the existing boards. 

Some of the fields are obvious, but I am unsure of others, e.g. upload.ram.maximum_size and the compiler flags. The F401 runs at 84MHz, and on the Nucleo F401RE board has 512 Kbytes flash and 96 Kbytes RAM. Any hints you could give me would be much appreciated.

p.s. the web based mbed IDE is okay, despite OS X complaining when you drag the .bin file to the board’s folder. I still need to figure out how to clone the tool chain locally so as to allow easy synch to github.  However, the Arduino tool chain is attractive in that makes it easier to share code with the AVR. My aim is to provide a portable open source web of things server that makes it easy for people to create their own distributed IoT applications.

madias123, 2015-08-26:
I think the forum - http://stm32duino.com/ is the better place for those questions. There is also a STM32F4 sub-board. Sadly the F4 line has only limited support (yet). 
The OSX warnings with drag&drop are really nasty only solution is to disable the system messenger (not a good idea)

rogerclarkmelbourne, 2015-08-26:
Dave,

Like Matthias has said, the information about this is on the forum

The F4 core is quite a lot different. It has the same parent as the F1, but its an older child / branch / copy, that didnt benefit from ore recent changes to later work that Leaflabs did on LibMaple 

So although we ( mainly Martin) have made quite a lot of changes in an attempt to update the F4 core, its still really somehwat of problem child

Hence why people like @sheepdoll have ditched it, and have started a new F4 core based on the STM CubeMX generated HAL files.

There is a growing head of steam to get in with moving the sheepdoll core from a technology demonstration / pre alpha ...  to something more usable / full featured, but the Northern hemisphere summer means is a quiet time for this community as everyone has more pressing things they need to do.

Anyway.

Like matthias said.

Please post to the forum, as its an easier way to discuss ideas

rogerclarkmelbourne, 2015-11-20:
Closed as this is an issue for the forum

closed 2015-11-20 by rogerclarkmelbourne