rogerclarkmelbourne/Arduino_STM32 #104: platform.rewrite.keys.txt

lloyd-g, 2015-08-28:
on my system it keeps looking for arm-none-eabi-gcc in \bin but normal install is \usr\bin 
is this set in this file as when I change it don't work?
old.0.compiler.path={runtime.ide.path}/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/
new.0.compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/
 is it set somewhere else?

rogerclarkmelbourne, 2015-08-28:
umm
did you install the Due or the Zero in order to get the ARM compiler?

If you did, which version of the IDE are you using and also what OS.

As I've not had any reports of anything like this happening before.

lloyd-g, 2015-09-01:
I had the ARM compilers installed as I was doing command line and make file compilations for STM32f103 chips and it worked file as they were on path.
the OS is Ubuntu 14 Linux not sure of the kernel.
if I removed  plaform.rewerite keys stuff  would it take the compiler from Linux path?

rogerclarkmelbourne, 2015-09-01:
The Arduino IDE generates the path, and uses the "recipies" in the platform.txt files

It looks like the current version of the IDE has some path problems, I'd recommend you go back to an old stable version of the IDE like 1.6.4

rogerclarkmelbourne, 2015-10-27:
I think soon, we'll drop support for old IDE versions e.g. 1.6.2 and earlier and then I'll be able to tidy up platform.txt, which may get rid of this error for you

rogerclarkmelbourne, 2015-11-09:
This has now been fixed in the master branch

closed 2015-11-09 by rogerclarkmelbourne