rogerclarkmelbourne/Arduino_STM32 #114: typo in udev script

tormodvolden, 2015-09-11:
tools/linux/45-maple.rules is missing comma before SYMLINK.

rogerclarkmelbourne, 2015-10-22:
I'll get this checked by one of the linux guru's

Andy... can you check this..???

pingumacpenguin, 2015-10-22:
Hi,

ATTRS{idProduct}=="1001", ATTRS{idVendor}=="0110", MODE="664", GROUP="plugdev"
ATTRS{idProduct}=="1002", ATTRS{idVendor}=="0110", MODE="664", GROUP="plugdev"
ATTRS{idProduct}=="0003", ATTRS{idVendor}=="1eaf", MODE="664", GROUP="plugdev", SYMLINK+="maple"
ATTRS{idProduct}=="0004", ATTRS{idVendor}=="1eaf", MODE="664", GROUP="plugdev", SYMLINK+="maple"

Works for me.... but then again the way it was phrased before without the comas also appeared to work. 
What issue was it causing?

rogerclarkmelbourne, 2015-10-27:
@tormodvolden 

What linux distro are you running ??

tormodvolden, 2015-10-27:
man udev | less +/comma

closed 2016-07-18 by rogerclarkmelbourne