rogerclarkmelbourne/Arduino_STM32 #115: dfu-util sources out of date

tormodvolden, 2015-09-11:
In particular, they don't point to the home page at dfu-util.sourceforge.net

BTW, are there any patches? Why do you have a binary in the source repo?

rogerclarkmelbourne, 2015-10-22:
Sorry to the tardy reply, but I have been away on business

Re: dfu sources
I'll need to check

They were correct when I added them.  

Its really hard to keep track of when the sources move, as there isnt a way to get notifications of this.

I had some other issues where a submodule git project was unlinked because the authors took their site offline, so I had to upload a local copy of the sources in question into the repo and gave up trying to link to the original authors copy (assuming they put it online again somewhere else)

Re: Binaries in the repo

Windows users have to have exe's for the tools. Most people don't have all the different compilers you'd need to build all the exe's

Even for Linux and OSX users, quite a few users of the repo a not able to install compilers and run makefiles and do all the other things that are needed to build from source, e.g. clone git repos or even run scripts

Hence I gave up expecting people to compile DFU from source, after having a myriad of support issues over this, and just ended up compiling binaries and putting them into the repo.

BTW. The version of DFU Util for windows seems to be a special one built by Leadlabs in 2012 and I don't have the source.

Re: patches etc for DFU util
This is question for the people who maintain dfu util

closed 2015-10-22 by rogerclarkmelbourne

tormodvolden, 2015-10-27:
Hi Roger,

Of course, no need to excuse, we all know this is a voluntary community project. Also for not keeping track of sudden upstream changes - in this case the whole domain disappeared without any possibility of recovery so it was not even possible to leave "we have moved" notice - it left the responsibility with the maintainers to try to update the most important downstreams - what is being done :)

When it comes to having binaries in the "repo" for the convenience of users: First, we are talking about a source code repository, which is not meant to include binaries. Normally one would distribute "convenience" binaries in release tarballs/zip files.

Upstream dfu-util has always provided binaries so it seems natural to point users to them, also taking away your support burden. The problem with multiple suppliers of binaries is that almost all of them will spread outdated versions, and cause confusion for users and frustration for the developers.

Secondly, your distribution of Leaflabs binaries without available sources would be a breach of the GNU license, so that is not a nice situation either :p

> "Re: patches etc for DFU util
> This is question for the people who maintain dfu util"

Um, no. If you are sitting on patches for dfu-util it is customary that you offer these patches to the dfu-util maintainers. I am not sure what you mean here? Or are there not any code changes in your dfu-util tree?

Best regards,
Tormod