rogerclarkmelbourne/Arduino_STM32 #116: stm32flash tweaks needed?

tormodvolden, 2015-09-11:
The git log says "…is a special version built for Arduino STM32". What are the changes needed, and why haven't they been sent upstream to stm32flash.sourceforge.net ?

KenjutsuGH, 2015-09-12:
Please see this post:

http://stm32duino.com/viewtopic.php?f=21&t=376&start=10#p3804

It explains which changes were made to facilitate **stm32flash** working correctly under **OSX**.

tormodvolden, 2015-09-12:
Thanks! Yes, checking against device names doesn't seem very robust. I am not sure it is necessary either, we'll open the device and set tty parameters later so it will fail then if the user is doing something very wrong. I opened a bug report here https://sourceforge.net/p/stm32flash/tickets/75/

closed 2015-09-12 by tormodvolden

tormodvolden, 2015-10-27:
Can anyone here test if isatty() would work on MacOSX for these devices? Please see above mentioned bug report.

rogerclarkmelbourne, 2015-10-27:
@tormodvolden 

Can you post your question / request to the main forum which is www.stm32duino.com  as there are far more users on there, than actively look at the github repo