rogerclarkmelbourne/Arduino_STM32 #117: Warnings in compilation

pierrotm777, 2015-09-19:
Please , can you add into your boards.txt files the directive:
xxxx.build.board=xxxx

Please , can you update into your platform.txt files the directive:
compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/

Thanks for your work !!!

rogerclarkmelbourne, 2015-10-22:
Sorry for the tardy reply, but I've been away on business

I can change the board.txt, but I'm not sure about the compiler path.

The reason the compiler path being different was to support older versions of the IDE, I think pre 1.6.2 but I can't remember the exact version number.

I guess I should force everyone to update to at least 1.6.3 and use a compiler path that only works for newer versions of the IDE

rogerclarkmelbourne, 2015-10-26:
Re:
Please , can you add into your boards.txt files the directive:
xxxx.build.board=xxxx

Which boards don't contain this ?

rogerclarkmelbourne, 2015-10-27:
Can't replicate. 

Please re-open and specify which boards still cause this issue.

closed 2015-10-27 by rogerclarkmelbourne