rogerclarkmelbourne/Arduino_STM32 #120: STM32 Discovery F40X boards compile error

hoobs1210, 2015-10-13:
I get the following error when trying to compile STM32 Discovery F40X boards.
location of the directory is valid.

```
arm-none-eabi-g++: error: Files: No such file or directory
arm-none-eabi-g++: error: (x86)\Arduino\hardware\Arduino_STM32\STM32F4\cores\maple/libmaple/usbF4: No such file or directory
arm-none-eabi-g++: error: Files: No such file or directory
arm-none-eabi-g++: error: (x86)\Arduino\hardware\Arduino_STM32\STM32F4\cores\maple/libmaple/usbF4/STM32_USB_Device_Library/Core/inc: No such file or directory
arm-none-eabi-g++: error: Files: No such file or directory
arm-none-eabi-g++: error: (x86)\Arduino\hardware\Arduino_STM32\STM32F4\cores\maple/libmaple/usbF4/STM32_USB_Device_Library/Class/cdc/inc: No such file or directory
arm-none-eabi-g++: error: Files: No such file or directory
arm-none-eabi-g++: error: (x86)\Arduino\hardware\Arduino_STM32\STM32F4\cores\maple/libmaple/usbF4/STM32_USB_OTG_Driver/inc: No such file or directory
arm-none-eabi-g++: error: Files: No such file or directory
arm-none-eabi-g++: error: (x86)\Arduino\hardware\Arduino_STM32\STM32F4\cores\maple/libmaple/usbF4/VCP: No such file or directory
Error compiling.

```

rogerclarkmelbourne, 2015-10-13:
Did you install the SAM board (due or Zero) in the boards manager ?

hoobs1210, 2015-10-13:
both Due and Zero are installed. i'm using 1.6.5

agmuntianu, 2015-10-13:
install arduino in a folder without spaces, it does not like "Program Files", for example in : C:\Arduino

hoobs1210, 2015-10-13:
Problem has been solved. Thank you all.

closed 2015-10-13 by hoobs1210