rogerclarkmelbourne/Arduino_STM32 #121: Missing _BV definition

sweetlilmre, 2015-10-22:
Hardly a burning issue but _BV seems to be undefined in the default set of includes and I couldn't seem to find it elsewhere. I'm trying to maintain a 'portable' set of code between a Maple Mini and Arduino Nano platform and this is a minor inconvenience. As this is a fairly common macro, perhaps it should be included somewhere?

sweetlilmre, 2015-10-22:
More on this. Seems its a known Due / ARM etc. core issue as described here:
https://github.com/arduino/Arduino/issues/1834

No movement on this since mid 2014 :(

rogerclarkmelbourne, 2015-10-22:
What does it do ?

I we can probably add it into one of the headers.

We have had to add various other macro's etc, as the original codebase we use is based on Arduino 0022, and as there is no overall API spec for the Arduino API, we only tend to know about omissions when people try to use things.

phenom128, 2015-10-22:
from atmel
http://www.atmel.com/webdoc/AVRLibcReferenceManual/group__avr__sfr_1ga11643f271076024c395a93800b3d9546.html

#define _BV(bit) \
	(1 << (bit))

#include <avr/io.h>

rogerclarkmelbourne, 2015-10-22:
Can you test that

#define _BV(bit) \
	(1 << (bit)) 

Does do what Atmel claim it does

"Converts a bit number into a byte value."

I can't see the on an ARM platform that this is going to result in a byte value.

More like an int, which is 32 bits

If this is supposed to return a byte, then I suspect the macro would need to cast the result back to a byte

phenom128, 2015-10-23:
well it was atmel avr libc,
apols fuzzy thinking or lack thereof

is'nt it returning a bit mask that the user has to apply in some way?

srp

rogerclarkmelbourne, 2015-10-23:
Yes, The result of this could be used as a bit mask

I think there are some other macros that we already have that do the same thing, but I understand that if you have a lib that uses this macro that it needs to be in the core

BTW. Does you code include avr/io.h ? i.e do we need a file with that name (and we'd stick the single marco in it, as using the whole AVR header could be problematic and cause conflicts

phenom128, 2015-10-23:
if 'our' code already has this macro albeit by another name and that macro
is 'visible' within
'our' include path, then a simple wrapper in that file would that suffice?

stephen

rogerclarkmelbourne, 2015-10-23:
My mistake 

The macros we have, don't do this.

I was thinking of this set

#define digitalPinToPort(P)        ( PIN_MAP[P].gpio_device )
#define digitalPinToBitMask(P)     ( BIT(PIN_MAP[P].gpio_bit) )
#define portOutputRegister(port)   ( &(port->regs->ODR) )
#define portInputRegister(port)    ( &(port->regs->IDR) )

#define portSetRegister(pin)		( &(PIN_MAP[pin].gpio_device->regs->BSRR) )
#define portClearRegister(pin)		( &(PIN_MAP[pin].gpio_device->regs->BRR) )

#define portConfigRegister(pin)		( &(PIN_MAP[pin].gpio_device->regs->CRL) )

I'm not sure if @sweetlilmre needs to be able to include avr/io.h

This will dictate where the macro gets defined.

sweetlilmre, 2015-10-23:
No need to include avr/io.h in a 'standard' Arduino build e.g.:
```
void setup() {
	int two = _BV(1);
}

void loop() {
}
```

will work. I think the broader set of definitions supplied in <avr/sfr_defs.h> would be very useful, though the comments in https://github.com/arduino/Arduino/issues/1834 hint at some complexity.
I could have a go at implementation.

rogerclarkmelbourne, 2015-10-23:
I'm not sure how useful it would be to implement the other macros, as from the issue you linked to, they seem to be associated with direct register access to the AVR hardware, which is going to be different on any other hardware, (Due or STM32  or Teensy etc)

So you could be wasting you time porting those.

Actually, I suspect they will compile OK, but will be of little value.

BTW. What code do you have that uses the _BV value, as if this also does direct hardware access, aimed at the AVR hardware, it will not work on STM32 (or Due). So you may need to completely rewrite whatever it is that uses _BV

sweetlilmre, 2015-10-23:
The code uses _BV for pure bit value creation so there is no direct HW access.
Perhaps the other macros are not in scope, but the final comment with the construct
```
#define _VOLVAL(_v)  (*((volatile typeof(_v) *) & _v)) // force value to be volatile
#define bit_is_set(sfr, bit) (_VOLVAL(sfr) & _BV(bit))
```

Looks promising. At any rate, some header that is included by default that contained:
```
#ifndef _BV
	#define _BV(bit) (1 << (bit))
#endif
```

Would be great.

rogerclarkmelbourne, 2015-10-23:
Probably put it in   cores\maple\wirish.h

Its already got these definitions which are similar

#define lowByte(w)                     ((w) & 0xFF)
#define highByte(w)                    (((w) >> 8) & 0xFF)
#define bitRead(value, bit)            (((value) >> (bit)) & 0x01)
#define bitSet(value, bit)             ((value) |= (1UL << (bit)))
#define bitClear(value, bit)           ((value) &= ~(1UL << (bit)))
#define bitWrite(value, bit, bitvalue) (bitvalue ? bitSet(value, bit) : \
                                                   bitClear(value, bit))
#define bit(b)                         (1UL << (b))

in fact the last one does the same job I think. Just by a different name

rogerclarkmelbourne, 2015-10-27:
Do you want em to add 

#ifndef _BV
    #define _BV(bit) (1 << (bit))
#endif 

to wirish,h ?

stevstrong, 2015-10-27:
I think it would make sense to add it in order to maintain Arduino compatibility.

sweetlilmre, 2015-10-27:
I would second that :)

rogerclarkmelbourne, 2015-10-27:
OK. 
Thats done.

closed 2015-10-27 by rogerclarkmelbourne