rogerclarkmelbourne/Arduino_STM32 #124: linux maple_upload.sh script fails if path contains a space

Quantalume, 2015-10-28:
Here is an example output:
```bash
/home/username/SpiderOak Hive/Arduino/sketchbook/hardware/Arduino_STM32/tools/linux/maple_upload: line 29: /home/username/SpiderOak: No such file or directory

/home/username/SpiderOak Hive/Arduino/sketchbook/hardware/Arduino_STM32/tools/linux/maple_upload: line 35: [: /home/username/SpiderOak: binary operator expected

/home/username/SpiderOak Hive/Arduino/sketchbook/hardware/Arduino_STM32/tools/linux/maple_upload: line 40: /home/username/SpiderOak: No such file or directory
```

There are possibly other scripts which would fail as I haven't tested them all.

rogerclarkmelbourne, 2015-10-29:
OK.

Thanks. I'll ask one of the community to do a fix

rogerclarkmelbourne, 2015-10-29:
I've pushed a fix, can you retest.

Quantalume, 2015-10-30:
Does not work. It looks like the Arduino IDE is passing the path, complete with spaces and lacking quotes, to maple_upload.  Maple_upload is then unable to tell the difference between a parameter with embedded spaces and an additional parameter. The issue needs to be sent upstream to the core Arduino developers.

By the way, I think the correct patch would be:
```
dummy_port=\"$1\"; altID=\"$2\"; usbID=\"$3\"; binfile=\"$4\"; dummy_port_fullpath="/dev/$1"
```
as the quotes need to be escaped.

Quantalume, 2015-10-30:
Disregard my comments about the Arduino IDE. The problem seems to stem from line 20 of maple_upload:
```
DIR=$( cd "$( dirname "${BASH_SOURCE[0]}" )" && pwd )
```
I've been unable to find a way to rewrite this to work in the case of a script calling a script in a path with spaces. Tried a lot of different things, including most everything from http://stackoverflow.com/questions/59895/can-a-bash-script-tell-what-directory-its-stored-in.

Quantalume, 2015-10-30:
Found something that works. Would be a good idea to make sure it doesn't break someone else's setup before merging. Thanks.

rogerclarkmelbourne, 2015-10-31:
OK.

I'll see if @ahull on the stm32 forum can test this, as he uses linux as his main platform

rogerclarkmelbourne, 2016-03-04:
Was it fixed by the merge on 31st Oct ?

Or does the problem still exist ?

closed 2016-07-18 by rogerclarkmelbourne