rogerclarkmelbourne/Arduino_STM32 #134: suggestion: sidebar for the wiki

jcw, 2015-11-01:
The righthand side of the wiki lists all the pages, but there are already enough that some of them drop off the main list. There's a feature in GitHub wikis to define a sidebar, with selected links and structure. This might be useful to make sure the presentation order is suitable for newcomers and as reference.

phenom128, 2015-11-01:
you might want to take a look at the chibios forum, same style forum, same
forum software phpbb

their quick links are visible, 'ours' are in the button 'quick links'
i loved indirection in 6502 assembler

stephen

jcw, 2015-11-01:
Sorry for the confusion, I meant the [wiki](https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki) on GitHub. It seems like a good place to point people to w.r.t. guides, tutorials, and reference material - in addition to the forum discussion that is.

> i loved indirection in 6502 assembler

:-)

rogerclarkmelbourne, 2015-11-01:
I think we need to do what the esp8266 github wiki does

https://github.com/esp8266/esp8266-wiki/wiki

Think in this case they are actually using files to link to pages in the wiki, but I'd need to clone the repo and take a look to understand how they do it

jcw, 2015-11-01:
Found it, I think: https://help.github.com/articles/creating-a-sidebar/

rogerclarkmelbourne, 2015-11-01:
OK.

Best to test this in a separate repo ;-)

rogerclarkmelbourne, 2016-06-16:
I have now created a separate wiki   wiki.stm32duino.com

closed 2016-06-16 by rogerclarkmelbourne