rogerclarkmelbourne/Arduino_STM32 #141: I get LOTS of errors from IDE 1.6.5 on a Due

louwin, 2015-11-15:
Absolute Newbie here so it is very probably MY fault.... 

It is just a fractal sketch I downloaded and am playing with (NOT my own work), just curious what it would look like on an LCD 320x240.

Running on Windows 10 Pro NOT Windows 8.1

Copied errors....

Arduino: 1.6.5 (Windows 8.1), Board: "Arduino Due (Programming Port)"

Build options changed, rebuilding all

In file included from DueFractal.ino:11:0:
C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due_gText.h:58:3: error: conflicting declaration 'typedef enum gTextFontMode gTextFontMode'
 } gTextFontMode;
   ^
In file included from C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due_gText.h:36:0,
                 from DueFractal.ino:11:
C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due.h:311:3: error: 'gTextFontMode' has a previous declaration as 'typedef enum gTextFontMode gTextFontMode'
 } gTextFontMode;
   ^
In file included from DueFractal.ino:11:0:
C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due_gText.h:155:2: error: redeclaration of 'gTextAlignTopLeft'
  gTextAlignTopLeft,
  ^
In file included from C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due_gText.h:36:0,
                 from DueFractal.ino:11:
C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due.h:318:2: note: previous declaration 'gTextAlign gTextAlignTopLeft'
  gTextAlignTopLeft,
  ^
In file included from DueFractal.ino:11:0:
C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due_gText.h:156:2: error: redeclaration of 'gTextAlignTopCenter'
  gTextAlignTopCenter,
  ^
In file included from C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due_gText.h:36:0,
                 from DueFractal.ino:11:
C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due.h:319:2: note: previous declaration 'gTextAlign gTextAlignTopCenter'
  gTextAlignTopCenter,
  ^
In file included from DueFractal.ino:11:0:
C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due_gText.h:157:2: error: redeclaration of 'gTextAlignTopRight'
  gTextAlignTopRight,
  ^
In file included from C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due_gText.h:36:0,
                 from DueFractal.ino:11:
C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due.h:320:2: note: previous declaration 'gTextAlign gTextAlignTopRight'
  gTextAlignTopRight,
  ^
In file included from DueFractal.ino:11:0:
C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due_gText.h:158:2: error: redeclaration of 'gTextAlignMiddleLeft'
  gTextAlignMiddleLeft,
  ^
In file included from C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due_gText.h:36:0,
                 from DueFractal.ino:11:
C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due.h:321:2: note: previous declaration 'gTextAlign gTextAlignMiddleLeft'
  gTextAlignMiddleLeft,
  ^
In file included from DueFractal.ino:11:0:
C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due_gText.h:159:2: error: redeclaration of 'gTextAlignMiddleCenter'
  gTextAlignMiddleCenter,
  ^
In file included from C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due_gText.h:36:0,
                 from DueFractal.ino:11:
C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due.h:322:2: note: previous declaration 'gTextAlign gTextAlignMiddleCenter'
  gTextAlignMiddleCenter,
  ^
In file included from DueFractal.ino:11:0:
C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due_gText.h:160:2: error: redeclaration of 'gTextAlignMiddleRight'
  gTextAlignMiddleRight,
  ^
[... 156 lines cut ...]
  gTextEraseFromBOL = 0x02, /**< Erase From Begining of Line to Cursor*/
  ^
In file included from DueFractal.ino:11:0:
C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due_gText.h:198:23: error: redeclaration of 'gTextEraseFullLine'
  gTextEraseFullLine = 0x03 /**< Erase Entire line */
                       ^
In file included from C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due_gText.h:36:0,
                 from DueFractal.ino:11:
C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due.h:345:2: note: previous declaration 'gTextEraseLine gTextEraseFullLine'
  gTextEraseFullLine = 0x03 /**< Erase Entire line */
  ^
In file included from DueFractal.ino:11:0:
C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due_gText.h:199:3: error: conflicting declaration 'typedef enum gTextEraseLine gTextEraseLine'
 } gTextEraseLine;
   ^
In file included from C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due_gText.h:36:0,
                 from DueFractal.ino:11:
C:\Program Files (x86)\Arduino\libraries\ILI9341_Due/ILI9341_due.h:346:3: error: 'gTextEraseLine' has a previous declaration as 'typedef enum gTextEraseLine gTextEraseLine'
 } gTextEraseLine;
   ^
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

rogerclarkmelbourne, 2015-11-16:
A couple of suggestions.

Support is generally on www.stm32duino.com not github

Also. I suspect the issue is in DueFractal.ino as it seems to re-use the same vars that are used in the lib.
 
Please post your query to www.stm32duino.com as the guys who wrote those libs don't read the github issues

Thanks

Roger

rogerclarkmelbourne, 2015-11-20:
Closed due to no feedback from OP

closed 2015-11-20 by rogerclarkmelbourne