rogerclarkmelbourne/Arduino_STM32 #142: Serial config parameter not working

stevstrong, 2015-11-18:
Hi,
I am using Serial2 on generic STM32F103C8T6. My application needs a signal with even parity.
Unfortunately, I can't get it working right.
In my sketch I am printing the ``udev->regs->CR1`` register value.
On the first try, I saw the bit PCE (bit 10 of CR1) for USART2 was not set at all.
I tracked down the issue and found the function ``usart_config_gpios_async`` in file ``usart_f1.c`` to be responsible to set this bit.
I saw that this function is called in`` HardwareSerial::begin(uint32 baud, unsigned config)`` in file ``HardwareSerial.cpp`` before the USART being initialized (before calling ``usart_init(this->usart_device)``).
Then, I changed the calling order, usart_init first and after that usart_config_gpios_async.
This way I can see in the sketch printout ``udev->regs->CR1`` register having the correct value (PCE bit set).
However, the output signal has still no parity bit.
What else I am missing here?

stevstrong, 2015-11-18:
OK, I got it.
The spec says:
``When the parity control is enabled, the computed parity is inserted at the MSB position (9th bit if M=1; 8th bit if M=0) and parity is checked on the received data.``
That means, if parity is wanted, bit M (word length) must be also set. This excludes a 9 bit length word with parity.
I think the header file should be adapted accordingly.
Should I send a PR?

stevstrong, 2015-11-18:
And another PR for changing the calling order of the mentioned functions in ``HardwareSerial.cpp``?

rogerclarkmelbourne, 2015-11-19:
Hi Steve,

I'll pull your changes into the development branch later today

rogerclarkmelbourne, 2015-11-20:
Should be pulled and pushed now.

closed 2016-07-18 by rogerclarkmelbourne