rogerclarkmelbourne/Arduino_STM32 #146: Initial push of RTClock gives missing 'maintainer' in 1.6.6 on windows

LjA1210113, 2015-12-03:
Error Message:
__Missing 'maintainer' from library in C:\Users\[name]\Documents\Arduino\hardware\Arduino_STM32-master\STM32F4\libraries\RTClock__

Windows 10 64bit, arduino IDE 1.6.6.
Other boards types F3/F1 compile without a hitch.
Thanks

LjA1210113, 2015-12-03:
Figured it out, 1.6.6 is fussy because the "maintainer=" field is missing from all the properties files in the F4 libraries....
Requesting pull request to fix it...
all compiles fine with them

rogerclarkmelbourne, 2015-12-04:
Yes
1.6.6 is fussy

It also has lots of other issues

I've given up chasing every new release of the IDE as often a new release has bugs

So currently 1.6.6 is not supported and may never be.

When things stabilize, in 1.6.7 etc we may be able to update the libs etc, but that will not be for some time

closed 2015-12-04 by rogerclarkmelbourne