rogerclarkmelbourne/Arduino_STM32 #147: Error compiling

sixtemesseven, 2015-12-03:
I followed your instructions but still get a compile error:

Arduino: 1.6.6 (Windows 10), Board: "Maple Mini, Bootloader 2.0 (20k RAM,120k Flash)"

Warning: platform.txt from core 'Arduino SAMD (32-bits ARM Cortex-M0+) Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Warning: platform.txt from core 'Arduino i586 Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Warning: platform.txt from core 'Arduino i686 Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Warning: platform.txt from core 'GD32 Boards (STM32duino.com)' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Warning: platform.txt from core 'STM32 Boards (STM32duino.com)' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Warning: platform.txt from core 'STM32F3 boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Warning: platform.txt from core 'STM32F4 boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
Board Intel:i586:izmir_fg doesn't define a 'build.board' preference. Auto-set to: I586_IZMIR_FG
Board Intel:i586:izmir_fd doesn't define a 'build.board' preference. Auto-set to: I586_IZMIR_FD
Board Intel:i686:izmir_ec doesn't define a 'build.board' preference. Auto-set to: I686_IZMIR_EC
WARNING: Category '' in library OneWireSTM is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library RTClock is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library Serasidis_EtherCard_STM is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library Serasidis_VS1003B_STM is not valid. Setting to 'Uncategorized'
exec: "/bin/arm-none-eabi-g++": file does not exist
Error compiling.
Invalid library found in C:\Users\sixtimesseven\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\libraries\Ethernet_STM: C:\Users\sixtimesseven\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\libraries\Ethernet_STM
Invalid library found in C:\Users\sixtimesseven\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\libraries\TFT_ILI9163C: C:\Users\sixtimesseven\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\libraries\TFT_ILI9163C
Invalid library found in C:\Users\sixtimesseven\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\libraries\Ethernet_STM: C:\Users\sixtimesseven\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\libraries\Ethernet_STM
Invalid library found in C:\Users\sixtimesseven\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\libraries\TFT_ILI9163C: C:\Users\sixtimesseven\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\libraries\TFT_ILI9163C
Invalid library found in C:\Users\sixtimesseven\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\libraries\Ethernet_STM: C:\Users\sixtimesseven\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\libraries\Ethernet_STM
Invalid library found in C:\Users\sixtimesseven\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\libraries\TFT_ILI9163C: C:\Users\sixtimesseven\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\libraries\TFT_ILI9163C

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

rogerclarkmelbourne, 2015-12-04:
Arduino 1.6.6 is not supported as its not stable

I may support 1.6.7 when its fully stable 

Please use 1.6.5

closed 2015-12-04 by rogerclarkmelbourne

sixtemesseven, 2015-12-04:
Ok, thanks for the quick answer! Great!!!

rogerclarkmelbourne, 2015-12-04:
No worries.

Sorry I can't do 1.6.6 at the moment but reading the Arduino forum etc, it seems 1.6.6 is causing loads of issues for other people, so there is no point wasting time on an unstable release.

sixtemesseven, 2015-12-04:
I agree. I wonder why they push out an unstable version on their main site for general use thought. Seems a bit silly to me.

rogerclarkmelbourne, 2015-12-04:
I think the community is in total agreement, and the IDE team seem to operate in a vacuum,  where they only consider Genuino products and built in libraries, but nothing else.

They don't seem to have a email notification system for library or "core" maintainers when they are going to release a new version which breaks existing libraries etc

I think if someone brought out an easy way to use the Arduino API and libraries using Eclipse, there would be a mass exodus from the Arduino IDE into Eclipse (or some similarly well supported IDE)

MauroMombelli, 2015-12-23:
can you please update the wiki to specify the arduino version to download?
i'm using 1.6.7 and i get 

    Warning: platform.txt from core 'STM32 Boards (STM32duino.com)' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.

and the it fail compiling because missing libmaple/stm32.h

downloading 1.6.5 right now

rogerclarkmelbourne, 2015-12-23:
Loads of changes were made in 1.6.6 which make it incompatible with files that worked will all previous versions (e.g. 1.6.5 or older)

Its not practical for me to maintain 2 different versions.

Additionally it wasnt just the core files that need updating; 1.6.6 is very picky about the libraries it works with, so a load of libraries would also need to be updated by their original authors before they would work for STM32 and IDE 1.6.6 

Hence for the moment its safer to stay with 1.6.5 and to wait for 1.6.7 etc and see what else the IDE team have decided to do.

Unfortunately, as far as I'm aware, the IDE dev team don't consult the community before adding features, or fundamentally changing the operation of the IDE :-(

MauroMombelli, 2015-12-23:
meanwhile i can confirm 1.6.5 works fine.

armleo, 2015-12-24:
1.6.5 working good can u specify what version to download on github page