rogerclarkmelbourne/Arduino_STM32 #149: Are the drivers working under Win10 64bit?

Protoneer, 2015-12-04:
I installed the drivers as specified but in my device manager I get the following device:

Unknown USB Device (Device Descriptor Request Failed)

I have these boards : http://www.aliexpress.com/item/5PCS-LOT-STM32F103C8T6-ARM-STM32-Minimum-System-Development-Board-Module-Forarduino/32355269970.html

I am not clear about what the jumpers on the board do, but I get the feeling they are needed to go into DFU mode.

stevstrong, 2015-12-04:
These kind of boards need the bootloader to be flashed, check the wiki how to do it for generic STM32 boards.

rogerclarkmelbourne, 2015-12-04:
Take a look on www.stm32duino.com as this is now the main community forum.

I read today on the forum that someone flashed using a Arduino UNO by holding the UNO in reset and connected to pins 0 and 1 to use its USB to Serial.

This seems a neat trick if it works as most people interested in Arduino have a Uno or similar

closed 2015-12-04 by rogerclarkmelbourne