rogerclarkmelbourne/Arduino_STM32 #156: Library does not compile with Arduino 1.6.7

derkostka, 2016-01-10:
I noticed, that with Arduino 1.6.7 (current release by now) some headers are not found:

Warning: platform.txt from core 'STM32 Boards (STM32duino.com)' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
In file included from /Users/sebastiankostka/Documents/Arduino/hardware/Arduino_STM32/STM32F1/cores/maple/Arduino.h:30:0,from sketch/Blink.ino.cpp:1:/Users/sebastiankostka/Documents/Arduino/hardware/Arduino_STM32/STM32F1/cores/maple/wirish.h:51:28: fatal error: libmaple/stm32.h: No such file or directory
 #include <libmaple/stm32.h>

darth-llamah, 2016-01-10:
AFAIK the latest supported Arduino IDE release is 1.6.5, 1.6.6 intruduced serious issues

SalvorinFex, 2016-01-10:
This subject has been brought up many times!
This is right on teh front page of this repo:
Summary:

```This repo contains, the "Hardware" files to support STM32 based boards on Arduino version 1.6.4 and 1.6.5 including LeafLabs Maple, and Maple mini, and other generic STM32F103 boards```

derkostka, 2016-01-10:
Yeah, i read this also. Nicht Problem everything is fine with 1.6.5, I just have been confused because I downloaded the linked arguing gui that resulted in the newest version.

I this cannot be fixed I am fine to close this task.

closed 2016-01-11 by rogerclarkmelbourne

gvuyk, 2016-01-11:
I didn't study the changes introduced on 1.6.6, 1.6.7 but got it working on 1.6.7 by moving files around so that stm32.h etc can be found. Everything seems fine so far (testing with generic STM32F103)

cstratton, 2016-05-19:
I believe it has something to do with a change to how the include paths are specified.

rogerclarkmelbourne, 2016-05-20:
The repo works with 1.6.9

The Arduino IDE team seem to have added in some backwards compatibility into 1.6.9 which makes this happen.

There is one warning if you use 1.6.9 but it does work.

Or if you don't want the warning, download the 'development' branch of the repo

I'm hoping that if 1.6.9 proves to be stable I will recommend everyone use that version until the next stable version if produced.