rogerclarkmelbourne/Arduino_STM32 #162: Very common functions are missing...

cunidev, 2016-02-12:
Hello,
I have noticed 3 missing functions that may cause incompatibility with Arduino sketches. They are:

tone(pin, frequency, [duration]) (the buzzer one...)
EEPROM.update(): The structure would be very simple: if(EEPROM.read(addr) != value) { EEPROM.write(addr, value)
EEEPROM.length(): could this be fixed by simply returning EEPROM.PageSize?

Thank you,
Raffaele (cunidev)

rogerclarkmelbourne, 2016-02-14:
If you can write the missing functions and submit a PR that would be great.

Thanks.

cunidev, 2016-02-14:
OK, I will as soon as I can.

cunidev, 2016-02-14:
Hmm... Just took a look, but the library seems really too big to allow me to add a function without the risk of breaking the entire software... What about if I wrote the code of the functions here in a comment directly, so that you know where to inject them?

rogerclarkmelbourne, 2016-06-16:
Any progress with your EEPROM lib changes?

cunidev, 2016-06-16:
Hello,
  I'm sorry, but I haven't been working with STM32 for months, so no
progress here :/
Anyway, the missing functions were almost aliases of existing functions, so
they could be added in 3 minutes or less to the library.

rogerclarkmelbourne, 2016-06-16:
No worries.

I'll leave the issue open as this library is part of the repo, so its probably best that the issue is documented and visible

rogerclarkmelbourne, 2016-12-03:
Closed due to inactivity

closed 2016-12-03 by rogerclarkmelbourne