rogerclarkmelbourne/Arduino_STM32 #164: u8glib with SSD1306 SPI display not compiling

cunidev, 2016-02-14:
I get these errors when trying to compile a u8glib-based sketch, could you help me? Thanks

`In file included from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/boards.h:38:0,
                 from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/wirish.h:53,
                 from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/Arduino.h:30,
                 from [Arduino]/libraries/U8glib/utility/u8g_com_arduino_common.c:46:
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/wirish_types.h:66:1: error: unknown type name 'bool'
 typedef bool boolean;
 ^
In file included from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/wirish.h:53:0,
                 from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/Arduino.h:30,
                 from [Arduino]/libraries/U8glib/utility/u8g_com_arduino_common.c:46:
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/boards.h:111:1: error: unknown type name 'bool'
 bool boardUsesPin(uint8 pin);
 ^
In file included from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/wirish.h:57:0,
                 from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/Arduino.h:30,
                 from [Arduino]/libraries/U8glib/utility/u8g_com_arduino_common.c:46:
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/ext_interrupts.h:90:6: error: conflicting types for 'attachInterrupt'
 void attachInterrupt(uint8 pin, voidArgumentFuncPtr handler, void *arg,
      ^
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/ext_interrupts.h:69:6: note: previous declaration of 'attachInterrupt' was here
 void attachInterrupt(uint8 pin, voidFuncPtr handler, ExtIntTriggerMode mode);
      ^
In file included from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/wirish.h:59:0,
                 from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/Arduino.h:30,
                 from [Arduino]/libraries/U8glib/utility/u8g_com_arduino_common.c:46:
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/wirish_math.h:58:6: error: conflicting types for 'random'
 long random(long min, long max);
      ^
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/wirish_math.h:49:6: note: previous declaration of 'random' was here
 long random(long max);
      ^
In file included from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/wirish.h:62:0,
                 from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/Arduino.h:30,
                 from [Arduino]/libraries/U8glib/utility/u8g_com_arduino_common.c:46:
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/wiring_pulse.h:34:72: error: expected ';', ',' or ')' before '=' token
 uint32_t pulseIn( uint32_t ulPin, uint32_t ulState, uint32_t ulTimeout = 1000000L ) ;
                                                                        ^
In file included from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/HardwareSerial.h:38:0,
                 from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/wirish.h:68,
                 from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/Arduino.h:30,
                 from [Arduino]/libraries/U8glib/utility/u8g_com_arduino_common.c:46:
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/Print.h:37:1: error: unknown type name 'class'
 class Print {
 ^
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/Print.h:37:13: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 class Print {
             ^
In file included from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/HardwareSerial.h:40:0,
                 from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/wirish.h:68,
                 from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/Arduino.h:30,
                 from [Arduino]/libraries/U8glib/utility/u8g_com_arduino_common.c:46:
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/Stream.h:38:1: error: unknown type name 'class'
 class Stream : public Print
 ^
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/Stream.h:38:14: error: expected '=', ',', ';', 'asm' or '__attribute__' before ':' token
 class Stream : public Print
              ^
In file included from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/wirish.h:68:0,
                 from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/Arduino.h:30,
                 from [Arduino]/libraries/U8glib/utility/u8g_com_arduino_common.c:46:
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/HardwareSerial.h:122:1: error: unknown type name 'class'
 class HardwareSerial : public Stream {
 ^
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/HardwareSerial.h:122:22: error: expected '=', ',', ';', 'asm' or '__attribute__' before ':' token
 class HardwareSerial : public Stream {
                      ^
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/HardwareSerial.h:183:2: error: unknown type name 'HardwareSerial'
  extern HardwareSerial Serial1;
  ^
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/HardwareSerial.h:186:2: error: unknown type name 'HardwareSerial'
  extern HardwareSerial Serial2;
  ^
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/HardwareSerial.h:189:2: error: unknown type name 'HardwareSerial'
  extern HardwareSerial Serial3;
  ^
In file included from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/wirish.h:69:0,
                 from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/Arduino.h:30,
                 from [Arduino]/libraries/U8glib/utility/u8g_com_arduino_common.c:46:
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/HardwareTimer.h:44:1: error: unknown type name 'class'
 class HardwareTimer {
 ^
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/HardwareTimer.h:44:21: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 class HardwareTimer {
                     ^
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/HardwareTimer.h:303:1: error: unknown type name 'HardwareTimer'
 extern HardwareTimer Timer1;
 ^
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/HardwareTimer.h:309:1: error: unknown type name 'HardwareTimer'
 extern HardwareTimer Timer2;
 ^
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/HardwareTimer.h:315:1: error: unknown type name 'HardwareTimer'
 extern HardwareTimer Timer3;
 ^
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/HardwareTimer.h:321:1: error: unknown type name 'HardwareTimer'
 extern HardwareTimer Timer4;
 ^
In file included from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/wirish.h:70:0,
                 from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/Arduino.h:30,
                 from [Arduino]/libraries/U8glib/utility/u8g_com_arduino_common.c:46:
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/usb_serial.h:41:1: error: unknown type name 'class'
 class USBSerial : public Stream {
 ^
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/usb_serial.h:41:17: error: expected '=', ',', ';', 'asm' or '__attribute__' before ':' token
 class USBSerial : public Stream {
                 ^
In file included from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/wirish.h:70:0,
                 from [Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/Arduino.h:30,
                 from [Arduino]/libraries/U8glib/utility/u8g_com_arduino_common.c:46:
[Arduino]/hardware/Arduino_STM32/STM32F1/cores/maple/usb_serial.h:77:2: error: unknown type name 'USBSerial'
  extern USBSerial Serial;
  ^`

I know it looks scary... :/ but I just don't get it. The board I'd like to compile it on is a STM32F103C8T6 board... But maybe you'd find it better to directly try to compile an u8glib "Hello World" example, which is likely to throw the same exact errors. The reason I am reporting this is that several sketches are based on U8glib, the most used display library in Arduino sketches.

Thanks.

cunidev, 2016-02-14:
BTW, as the compiler should be the same I don't think this matters, but I am working from a Mac

stevstrong, 2016-02-14:
Which Arduino version do you use?
Currently is only 1.6.5 supported.

cunidev, 2016-02-14:
I am on Arduino 1.6.5, but it still won't work :(

stevstrong, 2016-02-14:
I think I solved long time ago a similar issue by renaming the `u8g_com_arduino_common.c` file into .cpp.
Give it a try...

cunidev, 2016-02-14:
Hi,
Maybe that'd solve part of the problem, but in my case it continues
throwing the same errors (unknown type USBSerial etc.)... Thank you though.

rogerclarkmelbourne, 2016-02-14:
Which example were you trying to compile ?

dariomas, 2016-06-08:
Hello,
same problem here (the code can be compiled for Arduino Due but does not for Maple):
the graphics library U8glib, for Arduino is available in the Library Manager, is written in C with a C++ class wrapper: https://github.com/olikraus/u8glib
but the wirish code for libmaple is not "C compatible".

For reference try installing the U8glib library and compile the example HelloWorld.

rogerclarkmelbourne, 2016-06-09:
Most libraries need to be converted to work with libmaple

I think someone else may have investigated porting the library in question.

Try using google for search the stm32duino.com form e.g 
https://www.google.com/search?btnG=1&pws=0&q=u8glib+site:www.stm32duino.com

rogerclarkmelbourne, 2016-06-16:
As this is not an actual issue with the core (Its an issue with porting a library to a the core and the STM32 MCU), I will close this issue.

closed 2016-06-16 by rogerclarkmelbourne