rogerclarkmelbourne/Arduino_STM32 #165: Serial print where will it come out?

lloyd-g, 2016-02-23:
How do I work out where the serial will come out on any board? 
Got a STM32F103RGT6 chip so using STM32F103RE as a start. 
Does it go to USB or USART1?
Not getting anything at moment.

armleo, 2016-02-23:
It writes to USB.

ddrown, 2016-02-23:
The define SERIAL_USB will determine if the Serial object is the SerialUSB or UART1.  Your upload method determines if that preprocessor macro is defined - STM32duino bootloader and stlink turn it on, BMP and serial do not turn it on.

lloyd-g, 2016-02-24:
I'm using a stlink adaptor but I see nothing on USB. 
It has a USB_Disconnect line, 
[image] 
which controls 1k5 resistor USB_DP 
[image]
but switching this has no effect have and R79 is fitted?
Every thing hardware buzzes out.
lsusb on Ubuntu only sees my STLINK adaptor.

rogerclarkmelbourne, 2016-02-24:
If you want support, please use the forum www.stm32duino.com  

Its unclear whether your specific hardware is even supported

It looks like its got a variant of the Maple Mini USB reset hardware, except on the Maple mini, the USB disc pin is on PB9 

See https://github.com/leaflabs/maplemini/blob/master/maplemini.pdf

And in the repo, the Maple mini board type does not have STLInk defined as an upload type

What I suggest you do, is...

Make a new board variant that suits you board, by copying the generic F103V series, and then look at how the USB disconnect stuff is defined in the Maple Mini board and also in boards.txt

This is something you will need to do for yourself, as you are the person with this specific hardware

closed 2016-02-24 by rogerclarkmelbourne