rogerclarkmelbourne/Arduino_STM32 #169: Support STM32L053R8

probonopd, 2016-03-19:
It would be nice if the STM32L053 Nucleo-64 board with the STM32L053R8 MCU could be supported.

rogerclarkmelbourne, 2016-03-20:
I don't think supporting the L0 series devices would be that easy. From what I understand they are quite a bit different from the F103

So a substantial part of the low level stuff would need to be rewritten :-(

The F103 core uses code written my LeafLabs in 2012 (and earlier), for their Maple products.
At the time they released the core, STM had not made their Standard Peripheral Library / CMSIS open source, so LeafLabs had to write their own low level implementation and API.

Which means its quite hard to add new processor variants, as we can't simply replace the CMSIS files with the L series files 

There was some work done (my @sheepdoll) to rewrite the core using the CMSIS - as it got released as open source in mid 2015
See 

https://github.com/rogerclarkmelbourne/HALMX_Arduino_STM32

Which may be easier to migrate to the L0 series. 

But that core is only at Alpha stage, currently no  one seems to have the time to continue its development

probonopd, 2016-03-20:
Thanks for the explanation, I understand.

closed 2016-06-16 by rogerclarkmelbourne