rogerclarkmelbourne/Arduino_STM32 #173: ST-Link tools not working under Ubuntu 16.04 Linux?

WereCatf, 2016-04-15:
I only just received my generic STM32F103C8T6 and the ST-Link V2 - programmer and was eager to try it out, but.. alas, the ST-Link binaries fail when the Arduino IDE tries to execute them. Executing them manually from the command-line ain't faring any better, resulting in `bash: ./st-info: Tiedostoa tai hakemistoa ei ole` ie. "File or command not found"

Trying to peek at the libraries they been compiled against with ldd results in another error, saying something along the lines of "Executable file cannot be loaded dynamically"

What's goin' on here?

ddrown, 2016-04-16:
The Linux stlink commands are 32 bit, and I'm guessing you're using the 64 bit Ubuntu. There should be a way to install the 32 bit compatible libraries using Ubuntu's package manager.

WereCatf, 2016-04-16:
@ddrown Good point, doing `sudo apt-get install libusb-1.0:i386` fixed that. Alas, now there are new errors after a lot of "Flash page at addr whatever erased": ```
Flash page at addr: 0x08002c00 erased
Flash page at addr: 0x08003000 erased2016-04-16T03:11:06 ERROR src/stlink-common.c: flash loader run error
2016-04-16T03:11:06 ERROR src/stlink-common.c: run_flash_loader(0x8000000) failed! == -1
```

WereCatf, 2016-04-16:
Turns out it was some memory-protection thing at fault. I didn't see this mentioned anywhere, so it took me a good while to figure it out and fix it, and I still don't know if there is some better way of doing it than using the Windows-based ST Flash-utility.

closed 2016-04-17 by WereCatf