rogerclarkmelbourne/Arduino_STM32 #176: Compile Error

liborw, 2016-05-01:
When compiling blink:
```
Arduino: 1.6.8 (Mac OS X), Board: "Generic STM32F103C series, STM32F103C8 (20k RAM. 64k Flash), Serial"

WARNING: Category '' in library OneWireSTM is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library RTClock is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library SPI is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library Serasidis_EtherCard_STM is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library Serasidis_VS1003B_STM is not valid. Setting to 'Uncategorized'
WARNING: Category '' in library Serasidis_XPT2046_touch is not valid. Setting to 'Uncategorized'
Warning: platform.txt from core 'STM32 Boards (STM32duino.com)' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
In file included from /Users/libor/Documents/Arduino/hardware/Arduino_STM32/STM32F1/cores/maple/Arduino.h:30:0,
                 from sketch/Blink.ino.cpp:1:
/Users/libor/Documents/Arduino/hardware/Arduino_STM32/STM32F1/cores/maple/wirish.h:51:28: fatal error: libmaple/stm32.h: No such file or directory
 #include <libmaple/stm32.h>
                            ^
compilation terminated.
exit status 1
Error compiling for board Generic STM32F103C series.
```

rogerclarkmelbourne, 2016-05-01:
Only 1.6.4 and 1.6.5 are supported, due to big changes in 1.6.6 which are not backwards compatible, as indicated in the readme.

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/master/README.md 

I think @ddrown has produced a modified version that works with the Boards Manager that works with 1.6.6 and newer,

ddrown, 2016-05-01:
Install instructions for the 1.6.6+ compatible version are here: http://stm32duino.com/viewtopic.php?f=41&t=844

rogerclarkmelbourne, 2016-05-06:
Hi Dan

Now that it looks like 1.6.9 will work with the same platform.txt build recipe as 1.6.5, I'm going to try hosting a board manager package on the github stm32duino account.

I've make some changes to the repo today, and have removed the GD32 folder and made some minor changes to the STM32 to support the GD32

I've also removed some empty libraries, which were sub modules that were not normally getting pulled into the github download zip
And fixed up the problem that was causing the IDE to complain about the library category not being defined 
plus some other code fixes

I hope you don't mind, but initially, I'm using using your board manager files, but I'll to add a new Version with the changes I've just mentioned.

Actually, perhaps I should remove legacy versions you have, who switches from using your package  to the one I'm creating, will be forced to re-download everything again, and I'm not sure if its worth retaining all the old copies of the tools or cores (as you will still have them)

ddrown, 2016-05-07:
@rogerclarkmelbourne sounds great to me

rogerclarkmelbourne, 2016-05-07:
Hi Dan

No worries. I've almost completely rewritten you file, as I've removed the F3 and F4 and merged the GD32 into the STM32. 
Strangely the JSON file doesnt work with 1.6.5 but does work with the pre-release 1.6.9

I've also noticed that the ESP8266 have an automatic package generation script, (linux) so I'm going to see if that can be adapted to automatically create the JSON for the STM32, but I better get it working manually first ;-)

rogerclarkmelbourne, 2016-06-16:
This issue is now redundant, as the core works with 1.6.9

closed 2016-06-16 by rogerclarkmelbourne