rogerclarkmelbourne/Arduino_STM32 #179: Building a project fails with Arduino 1.6.8

neuhaus, 2016-05-30:
I installed a fresh copy of Android 1.6.8 and installed Arduino_STM32 per the instructions in the wiki: First installing Arduino SAM Boards 1.6.8 for Arduino Due support, then unpacking the ZIP file in the "hardware" folder and naming it Arduino_STM32.

I restarted the UI and I set the board to "Maple Mini".

When trying to compile a default empty project I get the following warnings and errors:

> Warning: platform.txt from core 'STM32 Boards (STM32duino.com)' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
> In file included from C:\Users\example\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/Arduino.h:30:0,
> 
>                  from sketch\sketch_may30a.ino.cpp:1:
> 
> C:\Users\example\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/wirish.h:52:28: fatal error: libmaple/stm32.h: No such file or directory
> 
>  #include <libmaple/stm32.h>
> 
>                             ^
> 
> compilation terminated.
> 
> exit status 1
> Error compiling for board Maple Mini.

For some reason Arduino doesn't find the include files.

neuhaus, 2016-05-30:
I guess it could work if I were to install Arduino 1.6.5, however I need a more recent version of Arduino for ESP8266 development.

Is there something I can edit in my platform.txt file to fix the include paths?

stevstrong, 2016-05-30:
Try the newest version, 1.6.9.

rogerclarkmelbourne, 2016-05-31:
I've uploaded the readme to show the repo seems to work OK with 1.6.9

(I'd not done this earlier as 1.6.9 was only released  a few weeks ago and we needed more feedback from users before updating docs, hence a lot of the docs still only refer to 1.6.5 )

neuhaus, 2016-06-07:
It works with 1.6.9.

I'm not sure I can use 1.6.8 because I also use the ESP8266 and in their [README](https://github.com/esp8266/Arduino) they advise to use 1.6.8.

What's required to make Arduino 1.6.8 work with STM32?

rogerclarkmelbourne, 2016-06-07:
I suspect the docs for ESP8266 are out of date and it would work fine with 1.6.9

Alternatively you can try the development branch of the repo
https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/development

In which I think I made some changes that may make it work with 1.6.8 - However, I never tested with 1.6.8 as there is really no need to use it since 1.6.9 now works fine.

Alternatively, there @ddrown  has done a boards manager package that works with 1.6.6, 1.6.7 and 1.6.8

But these versions are not officially supported by the repo.

closed 2016-06-08 by neuhaus