rogerclarkmelbourne/Arduino_STM32 #180: support for STM32F103T6

thomasfla, 2016-06-06:
Hello, To program a custom board based on STM32F103T6, I naively tried  to add this to board.txt:
```
## STM32F103T6 -------------------------
genericSTM32F103T.menu.device_variant.STM32F103T6=STM32F103T6 (6k RAM. 16k Flash)
genericSTM32F103T.menu.device_variant.STM32F103T6.build.cpu_flags=-DMCU_STM32F103T6
genericSTM32F103T.menu.device_variant.STM32F103T6.build.ldscript=ld/jtag.ld
genericSTM32F103T.menu.device_variant.STM32F103T6.upload.maximum_size=16384
genericSTM32F103T.menu.device_variant.STM32F103T6.upload.ram.maximum_size=6144
genericSTM32F103T.menu.device_variant.STM32F103T6.upload.flash.maximum_size=16384
```
but I got this error compiling a blink sketch: 

> series/stm32.h:255:2: error: #error "Bad STM32F1 configuration. Check STM32F1 <series/stm32.h> header."
>  #error "Bad STM32F1 configuration. Check STM32F1 <series/stm32.h> header."

I also tried to use existing settings like STM32F103T8 and upload it via serial link. Flashing is ok but the program doesn't start and the external crystal doesn't oscillate, even after hardware reset.

I think I must provide valid flash/ram address ? If so, where should I put it?  
Thank's, Thomas.

rogerclarkmelbourne, 2016-06-07:
If you add a new board, you need to also change

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/4f2a6028f07cfac35b3564263c57991e7eee7df0/STM32F1/system/libmaple/stm32f1/include/series/stm32.h

As there are some defines that need to be configured.

Re: Oscillator not running

Have you previously run any other code on the board to confirm you whether you have a hardware fault ?

The PLL settings are for a 8Mhz crystal, I don't know what would happen if you used a completely different crystal frequency. Its possible it may not oscillate at all

But if its 8Mhz, I would expect it to work

I'm not sure if its related, but I have one F103C8 board which will only run if you upload via serial, and then run the freshly uploaded code.
If I try to boot from cold, it doesn't run. I suspect it has a faulty crystal oscillator that takes ages to start, but as the board is a piece of junk, I didnt spend too long investigating what was going on.

But from what I remember, the core does not attempt to restart the oscillator if it fails to start. It just waits indefinitely for it to start.
With the board that I had problems with, I did attempt to add code to restart the osc after a timeout period, but it didn't fix the problem for me. So the board went back into the cupboard (where I keep all the defective boards)

I presume your hardware has the necessary external capacitors around the crystal ?

thomasfla, 2016-06-07:
It's alive!
Thank's a lot for your help. Yes I'm using a 8Mhz crystal with external capacitors.

After changing some defines, It now blink the LED and uses external crystal properly. 
For now, I use STM32_MEDIUM_DENSITY define (and it seems to work) since I do not clearly understand how to find STM32_NR_INTERRUPTS and STM32_TIMER_MASK values for the low density line.

If I do find the good values, and after some tests, are you interested in a PR?

rogerclarkmelbourne, 2016-06-07:
Yes. Send a PR if you find good values.

Actually.
You would need to create a new "variant" folder for this device, as some of the differences in parameters can't simply be set in boards.txt and stm32.h

Specifically, I took a quick look at the overview for the T6
http://www.st.com/content/st_com/en/products/microcontrollers/stm32-32-bit-arm-cortex-mcus/stm32f1-series/stm32f103/stm32f103t6.html

and the overview for the T8

http://www.st.com/content/st_com/en/products/microcontrollers/stm32-32-bit-arm-cortex-mcus/stm32f1-series/stm32f103/stm32f103t8.html

And the T6 only has 1 SPI (T8 has 2 SPI),
and the T6 only has 2 USARTS (T8 has 3 USARTS)

And as you know, the number of Timers is different

Its not that hard to create a new "variant". 

Just copy the existing STM32F103T folder to STM32F103T6 and then make the relevant changes in boards.txt)

If the device has less pins than the T8 device, you'd need to look at he PIN_MAP array boards.cpp (but looking at the PIN_MAP for the T8 I suspect that the T6 probably has the same number of GPIOS's as the T8 doesnt have that many)

BTW. 
Is this your own home made board or is it a commercial board. If its commercial, can you post a link, as I am building a wiki of boards that work on wiki.stm32duino.com

(You may also like to join the forum www.stm32duino.com as there may be someone else who uses this MCU - though I suspect its unlikely)

Incidentally, I ordered some T8 based boards a few days ago from a company in Hong Kong, so I will be able to confirm whether the T8 board variant works (I'm pretty sure it does as it was created by @jcw who runs Jeelabs)

rogerclarkmelbourne, 2016-06-16:
Any progress with your development ?

thomasfla, 2016-06-17:
The I2C, and UART works well, I still have some problem with USB, I declared what's needed for SerialUSB, the computer detect a new device but fails during listing.
To answer your previous comment, It is indeed a home made board (very specialized).
I think I will change my IC to a T8 with the same footprint, since I would like to use your boot-loader, and haven't too much time to adapt it.

Also, If we want to add support for the LOW_DENSITY devices, there will be a lot of duplicated code since it exist for STM32F103**x**4 and STM32F103**x**6. So maybe the hierarchy of boards should be modified?  What do you think?

rogerclarkmelbourne, 2016-06-17:
Have you tried the bootloader on you T6 device ? It may work. I have run it on an STLink dongle which has a very low spec STM321xx device (I can't remember which)

I'm not sure about changed needed for LOW_DENSITY, I'd need to look at the code again.

The problem with removing duplication, is that its hard to test the changes. So duplication is often the safer option.

rogerclarkmelbourne, 2016-07-18:
I will close this issue as its no longer active, and I don't have any T6 devices which I could use to add T6 support

closed 2016-07-18 by rogerclarkmelbourne