rogerclarkmelbourne/Arduino_STM32 #182: how to upload arduino code to stm32f401re

zamanturja, 2016-06-16:
there is no board manager for nucleo stm32f401re in arduino ide, only nucleo stm32f103rb availabe, what to do for it ?

rogerclarkmelbourne, 2016-06-16:
The boards manager package is maintained by @ddrown and is the secondary distribution method at the moment.

Most people install, by cloning the git repo or downloading the zip into their Arduino/hardware folder

Re:Nucleo F401E

As this is a community project and is not supported in any way by STM, boards are only added to the repo when a variant for that board is developed by a community member.

I don't know the differences between the F401 and the F405 / F407, but you may be able to use the variant files for one of those boards, as the basis for you development of a F401 core.

You may also like to look at Avik De's "Koduino" STM32 core, but that repo does not seem to support the F401 MCU either  (see https://github.com/avikde/koduino )

You could also take a look at the STM32L4 being developed by @GrumpyOldPizza

See, https://github.com/GrumpyOldPizza/arduino-STM32L4

As you may be able to modify that core to suit your needs

GrumpyOldPizza, 2016-06-16:
STM32L4 is a bit different than STM32F401. Different DMA engine, different
I2C engine. RX-Timeout is missing on F401. So there would be some work
involved.

I did use the Nucleo a lot for developement, and perhaps I'll add it one
day for official support. Really the kind of killer is that it does not
support USB (none of the Nucleo boards does).

My goal going forward was to add F446 support if that is of interest.

- Thomas

rogerclarkmelbourne, 2016-06-16:
@GrumpyOldPizza 

Thanks Thomas

I'm not personally that keen on either the Nucleo or Discovery range of boards from STM.
I find them a to restrictive.

One thing I failed to mention to the OP is that @sheepdoll's original HAL MX core did have the F401 as a variant. However, all boards except the F103 are currently commented out of boards.txt because vassilis only had time to work on one variant, when adding SPI and Serial CDC and all the other variants still need to be updated.
Hence I felt it best if the other, non working variants, were not visible at the moment.

GrumpyOldPizza, 2016-06-17:
I kind of like the Nucleo boards for quick and dirty prototyping. The
boards are cheap enough so that they are disposable. But they are not
really comparable to Arduino or say Teensy.

With the HAL MX core, it's always wise to limit the scope as much as
possible during initial development. No point in wasting time supporting
everybody's favourite board just to get bogged down in wild goose chases.
The  very same reason why the STM32L4 core officially only supports
Dragonfly, rather than DISCOVERY & Nucleo as well.

rogerclarkmelbourne, 2016-06-17:
Its always a trade off.
Getting code to work just for one scenario, when you know it needs to work in multiple scenarios, can lead to a lot of re-working.

But, with community development, getting enough momentum, (people involved), requires things to be working well enough for people to want to get involved.

Which is why in the HAL MX case, I thought it was better to ask Vassilis to focus on the F103C as this is a board that loads of people have.

Undoubtedly, the F103 core will need to be re-worked when we eventually get around to looking at the F4 or F3 or F0, but it can't be helped ;-)

Trying to write a core that will work with F0 to F7 is probably over optimistic, but a core that uses the CMSIS and HAL seems a logical way forward, so that existing code can be used.

However I have a niggling feeling that perhaps using libopencm3 may be a better base platform than the STM Cube, as at the moment the cube seems to be inconsistent in the code it generates for each series of MCU's, which makes it look like STM are really not allocating enough resources to its development and maintenance :-(

closed 2016-07-18 by rogerclarkmelbourne