rogerclarkmelbourne/Arduino_STM32 #185: Compile error: "exec: "/bin/arm-none-eabi-g++": file does not exist"

mettinoktay, 2016-07-05:
Downloaded and installed the files exactly as instructed in wiki, yet Arduino throws this error. As a solution, I downloaded gcc-arm compiler from Launchpad's website but nothing changed. Seems like a very fundamental problem that can be easily fixed with a simple Googling, but Google didn't help either.
Your recommendations will be appreciated.

RickKimball, 2016-07-05:
Are on windows? Do  you have a space in the path?

mettinoktay, 2016-07-05:
I'm on Windows and I don't have a space in the path.

RickKimball, 2016-07-05:
which version of the arduino ide?

RickKimball, 2016-07-05:
wait .. nm

RickKimball, 2016-07-05:
http://www.stm32duino.com/viewtopic.php?f=37&t=122&p=15537#p15537

mettinoktay, 2016-07-05:
Yes! I downloaded Arduino SAM Boards from IDE's board manager, and now it compiles. That post was surely a needle in the haystack. Thanks!

closed 2016-07-05 by mettinoktay

rogerclarkmelbourne, 2016-07-05:
Which Wiki

https://github.com/rogerclarkmelbourne/Arduino_STM32/wiki/Installation ??

First paragraph says the SAM board using the Boards manager

"Run the IDE, and on the Tools menu, select the Boards manager, and install the Arduino Due from the list of available boards"

Note. As Arduino.cc keep breaking things in the IDE, I need to update the wiki to freeze the version at 1.6.9 as 1.6.8 did work, but 1.6.5 did, and I have no idea if 1.6.10 (or whatever the call it) will work

RickKimball, 2016-07-06:
The wiki says install Due, my post on the stm32duino site says the Zero. Installing the Zero gets you the same arm-none-eabi-g++ toolchains as the Due, however it also installs openocd as a bonus.

rogerclarkmelbourne, 2016-07-06:
Thanks Rick

I'll change the Wiki (again), as I think  it was written prior to the release of the Zero

faelpinho, 2021-11-10:
I tested on Arduino 1.8.16 (latest), and still works.
To use lcd custom libs, we need to downgrade Adafruit GFX to 1.7.5 (maybe others libs as well)