rogerclarkmelbourne/Arduino_STM32 #188: Serial read with no data should return -1 not block

RickKimball, 2016-07-15:
The library reference for Serial.read says if no data is available it returns -1. Looking the arduino core it actually does this.  Seems like we should too no?

rogerclarkmelbourne, 2016-07-15:
Rick,

I totally agree.

I recall I had to change what some of the other Serial functions returned, but I didnt notice that descrepancy

RickKimball, 2016-07-16:
```
/*
 * I/O
 */

int HardwareSerial::read(void) {
    if (!this->available())
        return -1;

    return usart_getc(this->usart_device);
}
```

rogerclarkmelbourne, 2016-07-16:
We may as well bypass the call to this->available() can directly call
```
usart_data_available(this->usart_device);
```

as this is what available() does in the function just below the read()

```
uint32 HardwareSerial::available(void) {
    return usart_data_available(this->usart_device);
}
```

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/4f2a6028f07cfac35b3564263c57991e7eee7df0/STM32F3/cores/maple/wirish/HardwareSerial.cpp#LN128
(Line 128)

rogerclarkmelbourne, 2016-07-16:
I will change it to this, because usart_data_available looks like it returns the number of the bytes in the buffer, so testing against zero is probably more syntactically correct

```
int HardwareSerial::read(void) {
   
    if (usart_data_available(this->usart_device)==0)
	{
		return -1;
	}
 
    return usart_getc(this->usart_device);
}
```

(perhaps usart_data_available() should return a boolean but it returns an int

rogerclarkmelbourne, 2016-07-18:
It looks like this has already been fixed for the F4 using this code

```
int HardwareSerial::read(void) {
	if(usart_data_available(usart_device) > 0) {
		return usart_getc(usart_device);
	} else {
		return -1;
	}
}
```

So I'm going to use the same code on the F1 F3 and GD32 cores

rogerclarkmelbourne, 2016-07-18:
Fixed in

https://github.com/rogerclarkmelbourne/Arduino_STM32/commit/388e8efeb6bd556a190142624a5c7aeb9dbf97bb

closed 2016-07-18 by rogerclarkmelbourne