rogerclarkmelbourne/Arduino_STM32 #189: How to choose clock

Vitorbnc, 2016-07-18:
Hi,
I bought a generic STM32F103C dev board and it comes with a 8MHz crystal. How can I select the working clock of the board? Is it possible to set it to 72MHz?

berryelectronics, 2016-07-18:
The crystal doesn't set the frequency of the chip,  it's just there to give
a base clock which is getting divided inside the chip. So with a 8mhz
external crystal you get a 72mhz internal clock speed (which can be changed
somehow)

TL;DR: the number on the crystal doesn't represent the chip's speed

rogerclarkmelbourne, 2016-07-18:
@Vitorbnc 

This is a question not an issue.

Please register on www.stm32duino.com and post your question

closed 2016-07-18 by rogerclarkmelbourne

Vitorbnc, 2016-07-18:
Sorry for posting on the wrong place, but thanks for the answer!